isaac-sim/OmniIsaacGymEnvs #1: underwater environment

wangcongrobot, 2022-06-06:
Hi, thank you for the amazing project. Can we create an underwater environment for the robot in the Omniverse Gym? Thank you.

gavrielstate, 2022-06-06:
The physics simulation used by default in Isaac Sim and Omniverse does not include Hydrodynamics or Aerodynamics. If you're able to handle that aspect of the simulation yourself, you could apply the forces yourself. It looks like some of the docs for that kind of thing aren't up yet however.

wangcongrobot, 2022-08-25:
Thank you. According to some demos of Omniverse, there are some underwater scenes, like a pool. So if we create a USD world with a pool using Omniverse Create, can we use it for the robot? I notice that PhysX 5 has the fluid simulation function.

ganadevprajapathy, 2023-03-22:
@wangcongrobot Were you able to do fluid simulation as you expected on Isaac Sim?