isaac-sim/OmniIsaacGymEnvs #5: Domain Randomization

AlanSunHR, 2022-06-22:
Hi,

The standalone Isaac Gym provides a convenient domain randomization method by adding some parameters in config files. But for Omniverse Gym, I currently cannot find a method of doing so. Is there any workaround to achieve randomizations like adding noises to observations/states, and randomizing the friction/mass of an object?

Thanks!

kellyguo11, 2022-06-22:
Hi @AlanSunHR ,

We are working on adding a similar Domain Randomization framework in OmniIsaacGymEnvs, but it is not available in the current release. Adding noises to observations and states will have to be done through user code for now, and randomizing friction/mass properties have to go through the USD APIs. We will be working on adding new APIs on the physics side to allow setting these properties in the next release.

AlanSunHR, 2022-06-23:
Hi @kellyguo11,

Thanks for your quick response! Looking forward to the next release!

closed 2022-06-23 by AlanSunHR