isaac-sim/OmniIsaacGymEnvs #10: Calculation of `rand_delta` for ShadowHand and AllegroHand

j3soon, 2022-08-25:
In [`in_hand_manipulation.py`](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/d0eaf2e7f1e1e901d62e780392ca77843c08eb2c/omniisaacgymenvs/tasks/shared/in_hand_manipulation.py#L315), `rand_delta` is calculated as follows:

```py
rand_floats = torch_rand_float(-1.0, 1.0, (len(env_ids), self.num_hand_dofs * 2 + 5), device=self.device)
# ...
delta_max = self.hand_dof_upper_limits - self.hand_dof_default_pos
delta_min = self.hand_dof_lower_limits - self.hand_dof_default_pos
rand_delta = delta_min + (delta_max - delta_min) * rand_floats[:, 5:5+self.num_hand_dofs]
pos = self.hand_dof_default_pos + self.reset_dof_pos_noise * rand_delta
```

Let $L$ be `delta_min` and $U$ be `delta_max`. The line below maps `rand_floats` from $[-1,1]$ to $[2L-U,U]$:

```py
rand_delta = delta_min + (delta_max - delta_min) * rand_floats[:, 5:5+self.num_hand_dofs]
```

However, `rand_delta` should be in $[L,U]$ instead. If we want to map `rand_floats` from $[-1,1]$ to $[L,U]$, the code above should be changed into:

```py
rand_delta = delta_min + (delta_max - delta_min) * 0.5 * (rand_floats[:, 5:5+self.num_hand_dofs] + 1.0)
```

or change the assignment of `rand_floats` instead:

```py
rand_floats = torch_rand_float(0.0, 1.0, (len(env_ids), self.num_hand_dofs * 2 + 5), device=self.device)
```

The issue also exists in the [IsaacGymEnvs](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs) repo: [shadow_hand.py](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/isaacgymenvs/tasks/shadow_hand.py#L647) and [allegro_hand.py](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/isaacgymenvs/tasks/allegro_hand.py#L600).

kellyguo11, 2022-09-21:
Thanks for spotting this! We will have this fixed in our next release.

j3soon, 2022-12-29:
This is fixed in 61fe9d7e8710f6ca962fc4a3e44812f23970dc08 and https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/commit/26018690897814fda950ad7a00211f8999724a9d.

Thanks.

closed 2022-12-29 by j3soon