isaac-sim/OmniIsaacGymEnvs #11: raise Exception("Failed to create simulation view backend")

ArghyaChatterjee, 2022-09-19:
I was trying the demo run for anymal robot and got the following error:
```
rror executing job with overrides: ['task=AnymalTerrain', 'num_envs=64', 'checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2022.1/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/anymal_terrain.pth']
Traceback (most recent call last):
  File "scripts/rlgames_play.py", line 67, in parse_hydra_configs
    task = initialize_demo(cfg_dict, env)
  File "/home/arghya/OmniIsaacGymEnvs/omniisaacgymenvs/utils/demo_util.py", line 46, in initialize_demo
    env.set_task(task=task, sim_params=sim_config.get_physics_params(), backend="torch", init_sim=init_sim)
  File "/home/arghya/OmniIsaacGymEnvs/omniisaacgymenvs/envs/vec_env_rlgames.py", line 51, in set_task
    super().set_task(task, backend, sim_params, init_sim)
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py", line 82, in set_task
    self._world.reset()
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.core/omni/isaac/core/world/world.py", line 281, in reset
    SimulationContext.reset(self, soft=soft)
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 423, in reset
    SimulationContext.initialize_physics(self)
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 403, in initialize_physics
    self._physics_sim_view = omni.physics.tensors.create_simulation_view(self.backend)
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extsPhysics/omni.physics.tensors-1.4.15-5.1/omni/physics/tensors/impl/api.py", line 11, in create_simulation_view
    raise Exception("Failed to create simulation view backend")
Exception: Failed to create simulation view backend

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/python.sh: line 40: 19204 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

Here is the full terminal output:
```
arghya@arghya-Pulse-GL66-12UEK:~/OmniIsaacGymEnvs/omniisaacgymenvs$ PYTHON_PATH scripts/rlgames_play.py task=AnymalTerrain num_envs=64 checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2022.1/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/anymal_terrain.pth 
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omniisaacgymenvs', 'omniisaacgymenvs.utils', 'omniisaacgymenvs.utils.hydra_cfg', 'omniisaacgymenvs.utils.hydra_cfg.hydra_utils', 'omniisaacgymenvs.utils.hydra_cfg.reformat', 'omniisaacgymenvs.utils.demo_util', 'omniisaacgymenvs.utils.config_utils', 'omniisaacgymenvs.utils.config_utils.path_utils', 'omniisaacgymenvs.envs', 'omniisaacgymenvs.envs.vec_env_rlgames', 'omni.isaac.gym', 'omni.isaac.gym.vec_env', 'omni.isaac.gym.vec_env.vec_env_base', 'omni.isaac.gym.vec_env.vec_env_mt', 'omniisaacgymenvs.scripts', 'omniisaacgymenvs.scripts.rlgames_train', 'omniisaacgymenvs.utils.rlgames', 'omniisaacgymenvs.utils.rlgames.rlgames_utils', 'omniisaacgymenvs.utils.task_util'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the fillowing args:  ['/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--ext-folder', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts', '--ext-folder', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['task=AnymalTerrain', 'num_envs=64', 'checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2022.1/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/anymal_terrain.pth']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/logs/Kit/Isaac-Sim/2022.1/kit_20220918_195809.log
2022-09-19 00:58:09 [4ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.drivesim.sensors.nv.lidar' or '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.drivesim.sensors.nv.lidar/config'
[0.146s] [ext: omni.stats-0.0.0] startup
[0.156s] [ext: omni.gpu_foundation-0.0.0] startup
2022-09-19 00:58:09 [155ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.5], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.162s] [ext: carb.windowing.plugins-1.0.0] startup
[0.168s] [ext: omni.assets.plugins-0.0.0] startup
[0.169s] [ext: omni.kit.renderer.init-0.0.0] startup
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

2022-09-19 00:58:10 [1,588ms] [Warning] [gpu.foundation.plugin] Raytracing is not supported on GPU: 1. Skipping this device.

|---------------------------------------------------------------------------------------------|
| Driver Version: 470.141.3     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3060 Laptop.. | Yes: 0 |     | 6144    MB | 10de      | 0          |
|     |                                  |        |     |            | 2520      | f67dad92.. |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Graphics (ADL GT2)      |        |     | 48011   MB | 8086      | 0          |
|     |                                  |        |     |            | 46a6      | 42f3bec9.. |
|=============================================================================================|
| OS: Linux arghya-Pulse-GL66-12UEK, Version: 5.15.0-46-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 12th Gen Intel(R) Core(TM) i7-12700H | Cores: Unknown | Logical: 20
|---------------------------------------------------------------------------------------------|
[... 548 lines cut ...]
Traceback (most recent call last):
  File "scripts/rlgames_play.py", line 67, in parse_hydra_configs
    task = initialize_demo(cfg_dict, env)
  File "/home/arghya/OmniIsaacGymEnvs/omniisaacgymenvs/utils/demo_util.py", line 46, in initialize_demo
    env.set_task(task=task, sim_params=sim_config.get_physics_params(), backend="torch", init_sim=init_sim)
  File "/home/arghya/OmniIsaacGymEnvs/omniisaacgymenvs/envs/vec_env_rlgames.py", line 51, in set_task
    super().set_task(task, backend, sim_params, init_sim)
[... cut ...]
  File "/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/extsPhysics/omni.physics.tensors-1.4.15-5.1/omni/physics/tensors/impl/api.py", line 11, in create_simulation_view
    raise Exception("Failed to create simulation view backend")
Exception: Failed to create simulation view backend

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/python.sh: line 40: 19204 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

Ane help is appreciated. TIA.

kellyguo11, 2022-09-21:
Hi @ArghyaChatterjee,

The error `PxgCudaDeviceMemoryAllocator fail to allocate memory 134217728 bytes!! Result = 2` seems to suggest your GPU might have ran out of memory. Could you try reducing the number of environments to see if that helps? The Anymal robots have higher fidelity meshes than Cartpole and Ant, which could take up more memory.

ArghyaChatterjee, 2022-09-22:
Hi, I have 6 gbs of GPU memory. I am using Nvidia 3060 RTX. If I reduce the number of envs, still the result is similar. This time, the 1st scene loads but still it crashes after that with number of envs = 2

```
arghya@arghya-Pulse-GL66-12UEK:~/OmniIsaacGymEnvs/omniisaacgymenvs$ $ PYTHON_PATH scripts/rlgameplay.py task=AnymalTerrain num_envs=2 checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2022.1/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/anymal_terrain.pth 
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omniisaacgymenvs', 'omniisaacgymenvs.utils', 'omniisaacgymenvs.utils.hydra_cfg', 'omniisaacgymenvs.utils.hydra_cfg.hydra_utils', 'omniisaacgymenvs.utils.hydra_cfg.reformat', 'omniisaacgymenvs.utils.demo_util', 'omniisaacgymenvs.utils.config_utils', 'omniisaacgymenvs.utils.config_utils.path_utils', 'omniisaacgymenvs.envs', 'omniisaacgymenvs.envs.vec_env_rlgames', 'omni.isaac.gym', 'omni.isaac.gym.vec_env', 'omni.isaac.gym.vec_env.vec_env_base', 'omni.isaac.gym.vec_env.vec_env_mt', 'omniisaacgymenvs.scripts', 'omniisaacgymenvs.scripts.rlgames_train', 'omniisaacgymenvs.utils.rlgames', 'omniisaacgymenvs.utils.rlgames.rlgames_utils', 'omniisaacgymenvs.utils.task_util'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the fillowing args:  ['/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--ext-folder', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts', '--ext-folder', '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['task=AnymalTerrain', 'num_envs=2', 'checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2022.1/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/anymal_terrain.pth']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/kit/logs/Kit/Isaac-Sim/2022.1/kit_20220921_201711.log
2022-09-22 01:17:11 [7ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.drivesim.sensors.nv.lidar' or '/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/exts/omni.drivesim.sensors.nv.lidar/config'
[0.267s] [ext: omni.stats-0.0.0] startup
[0.278s] [ext: omni.gpu_foundation-0.0.0] startup
2022-09-22 01:17:11 [275ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.5], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
[0.286s] [ext: carb.windowing.plugins-1.0.0] startup
[0.295s] [ext: omni.assets.plugins-0.0.0] startup
[0.296s] [ext: omni.kit.renderer.init-0.0.0] startup
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

2022-09-22 01:17:13 [1,782ms] [Warning] [gpu.foundation.plugin] Raytracing is not supported on GPU: 1. Skipping this device.

|---------------------------------------------------------------------------------------------|
| Driver Version: 470.141.3     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3060 Laptop.. | Yes: 0 |     | 6144    MB | 10de      | 0          |
|     |                                  |        |     |            | 2520      | f67dad92.. |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Graphics (ADL GT2)      |        |     | 48011   MB | 8086      | 0          |
[... cut ...]
    self.progress_buf[:] += 1
RuntimeError: CUDA error: an illegal memory access was encountered
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
/home/arghya/.local/share/ov/pkg/isaac_sim-2022.1.1/python.sh: line 40: 28585 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

kellyguo11, 2022-09-22:
I just noticed you are trying to run the Anymal Terrain environment. Unfortunately, the terrain mesh does take up quite a lot more memory than other environments, likely more than the 6gb available on a 3060.

One thing you could try is reducing the dimensions of the terrain by modifying the below parameters in the task config AnymalTerrain.yaml file:
```
    mapLength: 8.
    mapWidth: 8.
    numLevels: 10
    numTerrains: 20
```
This might help reduce the memory consumption slightly, but you may have to experiment with it a bit to fit it on a 3060.

ArghyaChatterjee, 2023-01-03:
It worked, thanks. closing the issue.

closed 2023-01-03 by ArghyaChatterjee