isaac-sim/OmniIsaacGymEnvs #13: How does Randomization on Material Properties Work?

AlanSunHR, 2022-10-06:
Hi everyone,

I want to domain randomize the dynamic and static friction coefficients between the agent and the ground. I follow the instruction (modified the config file, added codes needed for DR), but it seams that the domain randomization does not work.

Here's how I check if it works or not:
1. I create a rigid_prim_view for the meshes whose material properties should be changed.
2. I set the randomization parameters as follows:
rigid_prim_views:
      knee_view:
        material_properties:
          on_reset:
            num_buckets: 250
            operation: "scaling"
            distribution: "uniform"
            distribution_parameters: [[0.0, 0.0, 1], [0.05, 0.05, 1]]
3. I start up the simulation. According to the parameters I set, the friction between the agent and the ground should be very low, and the ground should be very "slippery". However, the agent still walks very normally.

So I want to know what is the correct way to randomize the material properties? Do I need to first assign a material to the meshes? Actually I tried to create and assign materials to the meshes, but when I started training, the friction parameters in the UI did not change at all (I enabled the interaction between the env and the UI);

And I noticed that in the config file, there's a parameter called "default_physics_material". I suppose if I did not create any material for my agent, this will be the material for every mesh. Is this correct?

kellyguo11, 2022-10-14:
Hi @AlanSunHR ,

Your setup looks correct, you can try verifying whether the parameters are being randomized by calling `knee_view._physics_view.get_material_properties()`. Values in the UI will not be modified when using the GPU pipeline.

AlanSunHR, 2022-10-29:
Hi @kellyguo11, thanks for the reply!

I called the function as you told, but I am still confused.

The output dimension of the material properties is (8, 32, 3). I have 8 parts in the rigid body view, and I check the source code, so i think the the first dimension is the number of parts, the second is the number of shapes, and the third is the material properties. 

I did not apply any physics materials to any of the parts in the view, so I think the "default_physics_material" will be assigned to all the parts in the view. Some of the parts did have the material properties the same as the "default_physics_material". However, I found some of them having 0 friction and restitution, which is pretty weird, and I don't know where the 0 is coming from.

I suppose for a single part, by default, all shapes in this part should have the same material properties. But one part in my view has mixed material properties (some shapes have 0 friction, but others have friction of 1), what could be the possible problem here?

And speaking about the randomization, only one part (the first prim in the view) is randomized. 

Thanks a lot in advance!

dhajnes, 2023-12-14:
Hi, any update on this from both sides? @AlanSunHR did you find out whats the problem with the randomization? I am looking into applying higher friction into only the feet to simulate rubber feet of a robot and I don't want the default to override this setting.