isaac-sim/OmniIsaacGymEnvs #14: DIfference against IsaacGymEnvs

tengyu-liu, 2022-10-10:
Hi, 

What is the difference between this repo and IsaacGymEnvs?

DanielTakeshi, 2023-03-05:
Thanks @kellyguo11 !
Just curious will there be any update to Isaac Gym itself even with the shift in development efforts? A number of us have found CPU-GPU differences that are affecting performance where they give vastly different performance with the same inputs and have been posting here
https://forums.developer.nvidia.com/c/agx-autonomous-machines/isaac/isaac-gym/322

For example
https://forums.developer.nvidia.com/t/inverse-kinematics-has-different-results-on-the-cpu-and-gpu-full-mwe-provided/235626/4

At this point should we assume that there won't be any more potential support for IsaacGym? We are trying to figure out whether it's worth the investment to continue with IsaacGym or if we have to change to IsaacSim.

DanielTakeshi, 2023-04-02:
Hi @kellyguo11 just curious if you had any updates about this? Thank you for any information you may wish to provide.

dibbla, 2023-06-06:
Hi @kellyguo11! Is there any recommendations on whether to use Preview 4 or this repo?

DanielTakeshi, 2023-06-29:
Hi @kellyguo11 just curious if you had updates on this. Thanks!

kellyguo11, 2023-06-30:
Hi there, most of our development efforts have shifted towards Omniverse Isaac Sim and OmniIsaacGymEnvs. We currently do not have a plan on the roadmap for a new release of Isaac Gym Preview, but we will continue to push out updates and fixes with Isaac Sim releases.