isaac-sim/OmniIsaacGymEnvs #18: Deformable physics simulation during training

mingj1125, 2022-11-17:
Hi,

Thanks for providing this fantastic repo.

I am trying to setup a training task with deformable objects, but I found out that the deformable physics properties of the objects is not simulated during the training. But when I export the scene and simulate it in the new window, it is working. I tried to create a use.file with the deformable physics properties already set and also tried to define the deformable physics in the function set_up_scene in the task python file, but both resulted in the same issue. 

Does the training environment actually support deformable physics simulation? If yes, how can I setup the correct files for that? If not, will this be considered in the future?

Thanks a lot!

kellyguo11, 2022-12-12:
Hi there, could you try setting `use_flatcache: False` in your task config file? Soft bodies may not work correctly with flatcache enabled.

mingj1125, 2022-12-16:
Hi! Thank you for the reply. I tried setting that, but then even the rigid body simulation is no longer being displayed on screen. So nothing moves anymore. However, if I look at the training information and reward output from training and inference, the scene is having different rewards and somehow being simulated.

KevinGalassi, 2023-04-21:
Any update on the manner? I think i have the same problem...