isaac-sim/OmniIsaacGymEnvs #19: Cant install on windows 10

ShadowVik, 2022-11-25:
PYTHON_PATH -m pip install -e .
this command doesnt work and it gives the following : The system cannot find the path specified.
even that I can check it with doskey /m and it gives me the correct path "doskey PYTHON_PATH=C:\Users\user\AppData\Local\ov\pkg\isaac_sim-*\python.bat $*" where I changed user to the correct path name on my PC

closed 2022-11-30 by ShadowVik

ARobbins67, 2023-03-31:
Having a similar issue to this. Did you find a solution?

gavrielstate, 2023-03-31:
You need to change the PYTHON_PATH to have your actual user name, not 'user'.

ARobbins67, 2023-03-31:
@gavrielstate Still having an issue. Here were the commands I made:

```
C:\Users\alexa\OneDrive\Documents>doskey /macros
python_path=C:\Users\alexa\AppData\Local\ov\pkg\isaac_sim-2022.2.1\python.bat

C:\Users\alexa\OneDrive\Documents>PYTHON_PATH -m pip install -e .
Python 3.7.13 (default, Mar 28 2022, 15:21:59) [MSC v.1912 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

Macros just there to show it's a valid file path. The second command doesn't actually install anything it just opens up the python environment. Any help would be greatly appreciated.

gavrielstate, 2023-03-31:
You're running the `pip install -e .` in your Documents folder. Try running it in the `omniisaacgymenvs`  folder that you cloned.

We may need to be more explicit about that in the instructions