isaac-sim/OmniIsaacGymEnvs #23: domain randomization seems CANNOT change rigid prims' material properties

NJ-WuSiyang, 2023-01-07:
Hi,

I have tried the provided domain randomization code for the shadow hand task. I want to apply noises to **object_view**'s material properties, especially **restitution**, on reset. I **randomize the restitution of objects in two ranges**, `[0.0, 0.1]` or `[0.9, 0.99]`, and launched two runs. However, **the behavior of objects dropping to the ground seems identical**.

More specifically, I have made the following modifications to the released code:
+ In the `domain_randomization/randomization_params/rigid_prim_views/object_view/material_properties/on_reset` of the ShadowHand task config:
(1) set the operation to `direct`;
(2) set the distribution parameters to `[[0.7, 1, 0.0], [1.3, 1, 0.1]]` or `[[0.7, 1, 0.9], [1.3, 1, 0.99]]`.
+ In the `task/shared/in_hand_manipulation.py`, I register a `GeometryPrimView` for the ground plane, and apply a physics material with restitution `0.5` to it, during `set_up_scene`.

I launched two runs with the two different distribution parameters (one with restitution in `[0.0, 0.1]`, and the other with restitution in `[0.9, 0.99]`). I printed `object_view._physics_view.get_material_properties()`, which is indeed some random value in `[0.0, 0.1]` or `[0.9, 0.99]`, but the behavior of the simulation environment seems identical for these two different settings of object restitutions. When objects drop to the ground from the hand, none of these two restitution ranges makes the object bounce more than the other.

Why are there identical behaviors under different restitution ranges? I have attached the modified code below. Is there something wrong with it?
[omniisaacgymenvs_restitution_dr.zip](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/files/10365910/omniisaacgymenvs_restitution_dr.zip)

kellyguo11, 2023-01-10:
Hi @NJ-WuSiyang 

Thanks for reporting this and sharing your code. We will investigate the issue.

NJ-WuSiyang, 2023-03-16:
> Hi @NJ-WuSiyang
> 
> Thanks for reporting this and sharing your code. We will investigate the issue.

Are there any updates?

kellyguo11, 2023-03-16:
This will be fixed in the upcoming release, which should be available in the next couple of days.

closed 2023-03-29 by NJ-WuSiyang