isaac-sim/OmniIsaacGymEnvs #27: [Error] Memory allocation error when running examples involving hand models

ADebor, 2023-03-17:
Hi,

I encounter an issue when trying to run examples you provide that involve hand models, *i.e.* either the Allegro hand of the Shadow one. I tried to run several different commands, but I get the same error every time:

`PxgCudaDeviceMemoryAllocator fail to allocate memory 2147483648 bytes!! Result = 2`

I tried to run `scripts/random_policy.py` and `scripts/rlgames_train.py`, using `num_envs=1` (and more, but I assume that 1 is the lighter choice in terms of memory), with `test=True` (inference mode) as well as `test=False` (training mode), and with `headless=True` as well as `headless=False`. For all these, I get the exact same error mentioned above (always the same number of bytes, *i.e.* 2 GB), followed by lots of PhysX errors and a `Exception: Failed to create simulation view backend` exception at the end, and a segfault. The whole output is quite large, but I could paste it here if needed.

I tried running other examples (Anymal, AnymalTerrain, Cartpole, etc.) and they happen to work, even for multiple envs (*e.g.* Anymal with 50 robots). Bad luck, I'm working with robotic hands...

I did monitor my GPU to see what was happening concerning memory, and the GPU memory usage goes to a bit more than 4GB (for 6GB of available memory) then the program crashes. I assume the problem comes from that, since 2GB is thus more than available.

I don't really know where these 2GB come from, and why this seems to only happen with the two hands. [This issue](https://forums.developer.nvidia.com/t/error-pxgcudadevicememoryallocator-fail-to-allocate-memory-2147483648-bytes-result-2/235374?) seems related to mine, if that can help.

I don't know if such an issue is expected considering my specs (see below), or if you experienced such problems on your end, but since I'm planning to use Isaac Gym to do RL with the shadow hand, I'd appreciate your opinion on this problem. Feel free to ask for more info, including full terminal output if needed.

Thank you in advance for your help!

---

My environment:
Commit: [1aaf354](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/commit/1aaf354d6a2bcfd54525ab183d03313956b5968b)
Isaac Sim Version: 2022.2.0
OS: Ubuntu 20.04
GPU: RTX 3060
CUDA: 12.0
GPU Driver: 525.85.05

kellyguo11, 2023-03-17:
Hi there, can you try increasing the GPU buffers defined in [this ](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/cfg/task/ShadowHand.yaml#L78)section of the config file?

ADebor, 2023-03-18:
Thanks very much for the quick response! I tried to increase several of them, and it seems that increasing `gpu_max_rigid_patch_count` to more than 64 MB indeed works. At 64 MB, the simulation runs with rendering (using the random policy and *e.g.* 128 envs) but I get lots of `Patch buffer overflow detected, please increase its size to at least x in the scene desc!` errors. Increasing it to more than 64 solves the issue.

Considering I'm a newbie when it comes to Isaac Sim (and simulation in general), GPUs, PhysX, etc., could you provide me with more info on why increasing this buffer solves my issue, or maybe tell me where I can find this information? Is this related to friction patches mentioned [here](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/RigidBodyDynamics.html)? Or to the `patchStreamCapacity` buffer mentioned [here](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/GPURigidBodies.html)?

kellyguo11, 2023-03-20:
The GPU buffer dimensions determine the size of pre-allocated buffers required for PhysX. When the buffers are not large enough for PhysX to compute contacts and collisions, it will often generate warnings or errors. You can find more details in the docs [here](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/GPURigidBodies.html#tuning).

closed 2023-03-20 by ADebor