isaac-sim/OmniIsaacGymEnvs #29: [Question] Effort control mode (instead of position)

ADebor, 2023-03-28:
Hi there,

I'm currently modifying the ShadowHand example you provide in order to control the hand using efforts rather than position control. To do that, I use the `set_joint_efforts` method on the hand view (instance of `ShadowHandView(ArticulationView)`) instead of setting position targets and using `set_joint_position_targets` as done in the provided example.

This *seems* to work (the hand does move when using a random policy), but since the [`get_applied_joint_efforts`](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.articulations.ArticulationView.get_applied_joint_efforts) method isn't working (apparently a known bug that will be fixed soon, or is fixed and the problem is on my side) and returns nothing but zero efforts in the joints, I cannot really investigate and check that everything is working properly. 

What I am curious about is the [`set_drive`](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/220d34c6b68d3f7518c4aa008ae009d13cc60c03/omniisaacgymenvs/robots/articulations/shadow_hand.py#L106) call in the `ShadowHand` class. This call sets the drive type to either linear or angular through `pxr.UsdPhysics.DriveAPI`, as well as the target position or velocity to 0.0. Is this some kind of initialization step required with usd, or is this related to some behind-the-scene mechanism underlying the control of the hand, not depending on the type of control I choose using the Python API (*e.g.* by using [`switch_control_mode`](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.controllers.ArticulationController.switch_control_mode)), and preventing me from doing effort control on the hand?

I apologize if this is a silly question, and don't hesitate to tell me if it shoud be asked on the dev forum rather than here. Thanks!

kellyguo11, 2023-04-04:
Hi there, the UsdPhysics.DriveAPI schema shouldn't be required for effort control. You can set both the damping and stiffness values on the drives to be 0 for effort control. They are used mainly for position/velocity target controls. You can find more details about it in the USD documentation [here](https://openusd.org/dev/api/class_usd_physics_drive_a_p_i.html).

ADebor, 2023-04-05:
Thanks a lot @kellyguo11!

closed 2023-04-05 by ADebor