isaac-sim/OmniIsaacGymEnvs #30: Syntax Error

ARobbins67, 2023-03-30:
Can't solve the following syntax error. Running on Windows 11 cmd. I set PYTHON_PATH in the python.bat environment but now it's not letting me run rlgames_train.py

```
>>> PYTHON_PATH scripts/rlgames_train.py task=Cartpole
  File "<stdin>", line 1
    PYTHON_PATH scripts/rlgames_train.py task=Cartpole
                      ^
SyntaxError: invalid syntax
```
Tried running this in and out of the python environment. Any help is greatly appreciated.

ADebor, 2023-04-04:
Hi, it seems like you're trying to run the script from within the Python interpreter directly, which is not how it should be ran. See [this topic](https://stackoverflow.com/questions/13961140/syntax-error-when-using-command-line-in-python).

kellyguo11, 2023-04-04:
Hi there, yes it appears from the error message that you are trying to run the command from within the Python interpreter. The command should be run directly from the command prompt (outside of the Python interpreter).

closed 2023-05-05 by ARobbins67