isaac-sim/OmniIsaacGymEnvs #31: [Question] Adding contact sensors to shadow hand's fingertips

ADebor, 2023-03-30:
Hi there,

### Context

As stated [here](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/29#issue-1644034258), I'm currently modifying the ShadowHand example you provide in order to control the hand using efforts rather than position control, and I'd like to train RL models to achieve this using **contact/force/tactile observations** rather than "visual" ones (fingertip positions, angles, velocities, etc.) as done in the OpenAI examples for instance. 

### Wanted

In order to simulate the pressure/tactile sensors of the real Shadow Hand, I want to attach a contact sensor (as defined [here](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.sensor/docs/index.html#module-omni.isaac.sensor.scripts.contact_sensor)) to all fingertips of all envs, in order to get contact data at each step of my RL loop.

### Tried

In the `set_up_scene`method of my task class, I tried to create such sensors using the `IsaacSensorCreateContactSensor` command with `self.default_zero_env_path + "/shadow_hand/robot0_xxdistal"` as the prim path (xx being replaced by ff, mf, ...), and I then used the `ContactSensorInterface` in `get_observations` to get data from these sensors. I observed nothing but zero measurements, so I tried to visualize my sensors in the simulator by setting the `visualize` flag to True when using the creation command, and I set the `color` parameter to some kind of red. Turns out the sensors are located way above the hand (in the picture below, see the red dots outlining the spherical sensors), and do not move although the fingers do, which is not wanted. 

[image]

When setting `use_flatcache` to False in the config file, the hand appears standing vertically, and does not move anymore. However, the sensors' location matches in this case the position of the fingertips of this vertically standing hand. 

I also tried to create a `ContactSensor` object without the mentioned command, setting the prim path as stated before, and adding the sensor to the scene, but this was unsuccessful.

Could someone help me creating contact sensors attached to the controlled fingers? Thanks very much in advance!

kellyguo11, 2023-04-04:
Hi there, we also provide a vectorized API for contact sensors to retrieve net contact forces of RigidPrimViews that you may find helpful. We have an example of this in the Anymal Terrain task, which can be found [here](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/tasks/anymal_terrain.py#L365) with the setup done [here](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/robots/articulations/views/anymal_view.py#L52). This should work with `use_flatcache` set to True (setting this to False in the GPU pipeline will disable rendering, which is why the hands don't move). More documentation on the contact sensors class can be found [here](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.prims.RigidContactView).

ADebor, 2023-04-05:
This indeed works, thanks a lot @kellyguo11.

closed 2023-04-05 by ADebor

kaijieshi7, 2024-02-05:
Hi, @kellyguo11  why does the contact force in the Anymal Terrain task return 3 values? How to only get one value, like pressure from a pressure sensor.