isaac-sim/OmniIsaacGymEnvs #33: crash with unexpected keyword argument 'enable_livestream' when trying to run 'cartpole'

Robokan, 2023-04-09:
**python scripts/rlgames_train.py task=Cartpole**

/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/extscache/omni.pip.torch-1_13_0-0.1.4+104.1.lx64/torch-1-13-0/torch/utils/tensorboard/__init__.py:5: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  tensorboard.__version__
/home/bizon/anaconda3/envs/issac_sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/bizon/anaconda3/envs/issac_sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
Error executing job with overrides: ['task=Cartpole']
Traceback (most recent call last):
  File "scripts/rlgames_train.py", line 98, in parse_hydra_configs
    env = VecEnvRLGames(headless=headless, sim_device=cfg.device_id, enable_livestream=cfg.enable_livestream, enable_viewport=enable_viewport)
**TypeError: __init__() got an unexpected keyword argument 'enable_livestream'**

**I am running in a conda enviornment which is why I am using 'python' to launch.**
**Any ideas? I am completely stuck here.**

kellyguo11, 2023-04-22:
Hi there, could you please make sure you are running with the latest Isaac Sim release, version 2022.2.1? The `enable_livestream` flag was introduced in the latest release.

Robokan, 2024-02-04:
Fixed this by ensuring to run the setup.py un the Cinda environment