isaac-sim/OmniIsaacGymEnvs #36: ModuleNotFoundError: No module named 'omni.isaac.franka'

nuomizai, 2023-04-19:
This error occurs when I tried to run the FrankaCabinet taks with headless mode. I didn't change the script under tasks/franka_cabinet.py. The command is as follows:

```
/home/midea/.local/share/ov/pkg/isaac_sim-*/python.sh omniisaacgymenvs/scripts/rlgames_train.py task=FrankaCabinet headless=True
```

Then I got:

```
[1.760s] [ext: omni.kit.window.title-1.1.2] startup
[1.761s] [ext: omni.isaac.gym-0.3.3] startup
[1.761s] [ext: omni.isaac.sim.python.gym.headless-2022.2.1] startup
[1.762s] Simulation App Starting
[2.268s] app ready
2023-04-19 12:38:12 [2,265ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/.cloudfront.toml
2023-04-19 12:38:12 [2,265ms] [Warning] [omni.client.plugin]  HTTP Client: omniclient: CC-873: Bypassing cache until the application is restarted
[2.284s] Simulation App Startup Complete
Setting seed: 42
Sim params does not have attribute:  physx
Sim params does not have attribute:  franka
Sim params does not have attribute:  cabinet
Sim params does not have attribute:  prop
Pipeline:  GPU
Pipeline Device:  cuda:0
Sim Device:  GPU
[2.309s] [ext: omni.inspect-1.0.1] startup
[2.316s] [ext: omni.kit.clipboard-1.0.0] startup
[2.320s] [ext: omni.kit.menu.create-1.0.8] startup
[2.327s] [ext: omni.volume-0.1.0] startup
[2.330s] [ext: omni.kit.context_menu-1.5.12] startup
[2.333s] [ext: omni.activity.core-1.0.1] startup
[2.335s] [ext: omni.hydra.rtx-0.1.0] startup
[2.342s] [ext: omni.debugdraw-0.1.1] startup
[2.348s] [ext: omni.kit.widget.stage-2.7.24] startup
[2.352s] [ext: omni.kit.window.property-1.8.2] startup
[2.354s] [ext: omni.kit.viewport.utility-1.0.14] startup
[2.355s] [ext: omni.kit.property.usd-3.18.17] startup
[2.362s] [ext: omni.kit.widget.text_editor-1.0.2] startup
[2.364s] [ext: omni.kit.widget.settings-1.0.1] startup
[2.365s] [ext: omni.kit.widget.graph-1.5.6-104_2] startup
[2.377s] [ext: omni.ui.scene-1.5.18] startup
[2.382s] [ext: omni.kit.window.preferences-1.3.8] startup
[2.480s] [ext: omni.kit.window.extensions-1.1.1] startup
[2.484s] [ext: omni.kit.widget.prompt-1.0.5] startup
[2.485s] [ext: omni.graph.ui-1.24.2] startup
[2.519s] [ext: omni.graph.scriptnode-0.10.0] startup
[2.521s] [ext: omni.graph.action-1.31.1] startup
[2.530s] [ext: omni.graph.bundle.action-1.3.0] startup
[2.530s] [ext: omni.syntheticdata-0.2.4] startup
2023-04-19 12:38:12 [2,527ms] [Warning] [omni.syntheticdata.scripts.extension] SyntheticData extension needs at least a stageFrameHistoryCount of 3
[2.546s] [ext: omni.command.usd-1.0.2] startup
[2.548s] [ext: omni.replicator.core-1.7.8] startup
2023-04-19 12:38:12 [2,553ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
[2.616s] [ext: omni.replicator.isaac-1.7.4] startup
Error executing job with overrides: ['task=FrankaCabinet', 'num_envs=1', 'headless=True']
2023-04-19 12:38:12 [2,822ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/
An error occurred during Hydra's exception formatting:
AssertionError()
Traceback (most recent call last):
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py", line 252, in run_and_report
    assert mdl is not None
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "omniisaacgymenvs/scripts/rlgames_train.py", line 143, in <module>
    parse_hydra_configs()
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/main.py", line 52, in decorated_main
    config_name=config_name,
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py", line 378, in _run_hydra
    lambda: hydra.run(
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py", line 381, in <lambda>
    overrides=args.overrides,
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "omniisaacgymenvs/scripts/rlgames_train.py", line 115, in parse_hydra_configs
    task = initialize_task(cfg_dict, env)
  File "/home/midea/wk/isaac_proj/OmniIsaacGymEnvs/omniisaacgymenvs/utils/task_util.py", line 41, in initialize_task
    from omniisaacgymenvs.tasks.franka_drawer import FrankaDrawerTask
  File "/home/midea/wk/isaac_proj/OmniIsaacGymEnvs/omniisaacgymenvs/tasks/franka_drawer.py", line 33, in <module>
    from omni.isaac.franka import Franka
ModuleNotFoundError: No module named 'omni.isaac.franka'
2023-04-19 12:38:13 [3,097ms] [Warning] [carb.audio.context] 1 contexts were leaked
/home/midea/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh: line 41: 10311 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

```

But this error gones when I set `headless=False`. How can I use headless mode with the task FrankaCabinet?

kellyguo11, 2023-04-22:
Hi there, the reason for the error is because in headless mode, we include a minimized set of extensions to reduce the time it takes to start up the process. To add the `omni.isaac.franka` extension to the set of extensions included, you can modify the app file in the Isaac Sim directory under `apps/omni.isaac.sim.python.gym.headless.kit`, by adding the line `"omni.isaac.franka" = {}` to the bottom of the file.

nuomizai, 2023-04-24:
> Hi there, the reason for the error is because in headless mode, we include a minimized set of extensions to reduce the time it takes to start up the process. To add the `omni.isaac.franka` extension [...]

Thanks for your reply, @kellyguo11 . Now it works well in the headless mode.

closed 2023-04-24 by nuomizai