isaac-sim/OmniIsaacGymEnvs #37: Different behaviour from isaac sim

KevinGalassi, 2023-04-21:
Hello,
I have a pick and place application, where a robot is grasping a cloth created using ClothPrim.
I started from isaac sim and i was able to grasp the cloth, and when i integrated everything inside this repository i have the following behavior. 

The cloth parameters are exactly the same and i use the same robot model as well. So i don't know exactly what to do...

[Screencast from 04-21-2023 12:02:09 PM.webm](https://user-images.githubusercontent.com/34882536/233644422-82d962a6-db37-49dd-985a-0f9d63208ed6.webm)

kellyguo11, 2023-04-22:
Hi there, it's possible that this repository is applying additional default values to some of the physics parameters for the simulation, robots, and objects. Many of the parameters can be adjusted from the config files. The simulation results may also vary depending on whether the GPU or CPU APIs are used. It may be a good start to compare the attributes for the physics scene and objects to see if they are the same when launched from OIGE and from Isaac Sim.

KevinGalassi, 2023-04-24:
I checked the parameters of the PhysicScene and they look the same, while i'm using the same objects definition in both the files, so i have no idea.
Another thing that i noticed that the debug of physics behave differently when using OIGE (e.g. particles are shown in spawn position and not follow the cloth representation)

That bring me think that the by default the two application spawned are different but i don't know

KevinGalassi, 2023-05-02:
Do you know how to change the setting so that i can obtain all USD as in isaac application instead of the output you can get from OIGE? It may be related to the difference behavior i'm experiencing.

At first, it seems that it was related to the flatcache setting, and if I set it to False, the cameras are blocked despite having my application running in the background

OIGE output :
[image]

Issac sim app:
[image]

KevinGalassi, 2023-05-09:
with pipeline=cpu i obtain the following error: 

2023-05-09 07:43:44 [22,619ms] [Error] [omni.physx.tensors.plugin] CpuSimulationView::createParticleClothView is not implemented yet

Other parameters that may be usefull are :
```
## Device config
physics_engine: 'physx'
# whether to use cpu or gpu pipeline
pipeline: 'cpu'
# whether to use cpu or gpu physx
sim_device: 'gpu'
```

I also tried disabling flatcahe, but if it is set to false the robot is not moving anymore while the program is running in the background