isaac-sim/OmniIsaacGymEnvs #38: Net Contact Forces is always Zero for Anymal Robot

AlanSunHR, 2023-04-22:
It seems that the contact forces of the Anymal's knees are not correct values, and in the task AnymalTerrain, the reset is not triggered while the knees contact with the terrain.

It is simple to reproduce the issue:
1. Change the defaultJointAngles in AnymalTerrain.yaml to:
    LF_HAA: 0.03    # [rad]
    LH_HAA: 0.03    # [rad]
    RF_HAA: -0.03   # [rad]
    RH_HAA: -0.03   # [rad]

    LF_HFE: 0.4     # [rad]
    LH_HFE: -0.4    # [rad]
    RF_HFE: 0.4     # [rad]
    RH_HFE: -0.4    # [rad]

    LF_KFE: -2.8    # [rad]
    LH_KFE: 2.8     # [rad]
    RF_KFE: -2.8    # [rad]
    RH_KFE: 2.8     # [rad]
2. Run the following command in the terminal:
python scripts/rlgames_train.py task=AnymalTerrain headless=false num_envs=2 test=True

We will see that the knees of the Anymal robots touches the ground but the env is not terminated, and I found that the returned value of self._anymals._knees.get_net_contact_forces(clone=False) is always 0.
[image]

AlanSunHR, 2023-04-22:
Update some of my observations here.

I think it might be the problem of the cylinder collision mesh. I found that the collision meshes of Anymal are all spheres except for the knees, which are cylinders.

I manually changed the one of the knees to sphere mesh, and the reading is normal for this specific knee. I also encountered some problems with cylinder meshes before, where the physics seems to be very abnormal if I have a cylinder mesh in my environments.

 And another problem is the way of calling the function self._anymals._knees.get_net_contact_forces(clone=False), I think we should add the argument dt=0.005? Otherwise the contact forces will be very small and the knee_contact reset will never be triggered.

kellyguo11, 2023-04-22:
Hi there, thanks for reporting the issue, we will investigate this. Could you please verify the version of Isaac Sim you are working with? We've made some fixes to the contact force API for the Isaac Sim 2022.2.1 release.

AlanSunHR, 2023-04-23:
Hi, I'm using Isaac Sim 2022.2.1 release.

closed 2023-04-23 by AlanSunHR