isaac-sim/OmniIsaacGymEnvs #40: How to generate videos of training ?

arminvburren, 2023-04-24:
Hello, I've been looking around the doc and examples for a while but I can't find an example. Is there a way to generate videos, either during training at some points, or at the end of training ?

I've seen that contrary to the original IsaacGym repo, the render "rgb" function has not been implemented here, but on the other hand there's a very recent commit called "d62b283 support headless render" which would be exactly what I want.

Even just outputing a single image would be fine, I could do the script later to make a video.

Thanks.

kellyguo11, 2023-05-01:
Hi there, we do not have an API to capture recordings from the viewport at the moment. For now, you can have a look at the Movie Capture tool (https://docs.omniverse.nvidia.com/prod_extensions/prod_extensions/ext_movie-capture.html), which can be used to capture images or videos of the viewport. We will look at adding this feature in the future.

arminvburren, 2023-06-04:
Hello, so i've tried to put this code in set_initial_camera_params in rl_task: 

```
            capture_filename = "capture_png_test"
            filePath = pathlib.Path("./").joinpath(capture_filename)
            options = CaptureOptions()
            options.file_type = ".png"
            options.output_folder = str(filePath)
            # make_sure_directory_existed(options.output_folder)
            # clean_files_in_directory(options.output_folder, ".png")
            exr_path = os.path.join(options._output_folder, "Capture1.png")
            # carb.log_warn(f"Capture image path: {exr_path}")
            options.hdr_output = False
            options.camera = viewport_api_2.camera_path.pathString

            capture_instance = CaptureExtension().get_instance()
            capture_instance.options = options
            capture_instance.start()
```

Basically it crashes by saying:

  File "/home/me/Classeur/OmniIsaacGymEnvs/omniisaacgymenvs/tasks/base/rl_task.py", line 161, in set_initial_camera_params
    capture_instance.options = options
AttributeError: 'NoneType' object has no attribute 'options'

So CaptureExtension() is None. After investigation, it may come from the fact that there's an extension missing. For example when i launch a OmniIsaacGym script, the only extension with name 'capture' is "[1.185s] [ext: omni.kit.renderer.capture-0.0.0] startup", whereas if I launch isaac sim there's at least these ones more:

[6.705s] [ext: omni.kit.window.movie_capture-2.1.4] startup
[6.651s] [ext: omni.kit.capture.viewport-1.1.20] startup

Do I need to add these extensions to the launch of OmniIsaacGym ? How do I do this concretely ? I've watched a bit and it's not clear at all to me ?

sambaPython24, 2023-09-06:
Are there any news on this?

arminvburren, 2023-09-06:
Well i found another way to doing it. What are you working on exactly ? Maybe we can share some effort

sujitvasanth, 2023-09-30:
I tried the movie capture extension but it isn't very friendly and produces frames or wrongly timed videos
its a poor solution but I record my videos using OBS studio during training and then run a python program I woite myself to remove the free\wa during learning https://github.com/sujitvasanth/VideoFrozenFramesRemover