isaac-sim/OmniIsaacGymEnvs #45: running torch.distributed.run on my dual 4090 GPU workstation crashes immediately when training the Ant example.

Robokan, 2023-05-05:
**# I am using a conda environment which has been set up properly as per instructions
python -m torch.distributed.run --nnodes=1 --nproc_per_node=2 scripts/rlgames_train.py headless=True task=Ant multi_gpu=True**

Any ideas? I would love to be able to use my other 4090!

stack trace:

python -m torch.distributed.run --nnodes=1 --nproc_per_node=2 scripts/rlgames_train.py headless=True task=Ant multi_gpu=True
WARNING:__main__:
*****************************************
Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
*****************************************
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omniisaacgymenvs', 'omniisaacgymenvs.utils', 'omniisaacgymenvs.utils.hydra_cfg', 'omniisaacgymenvs.utils.hydra_cfg.hydra_utils', 'omniisaacgymenvs.utils.hydra_cfg.reformat', 'omniisaacgymenvs.utils.rlgames', 'omniisaacgymenvs.utils.rlgames.rlgames_utils', 'omniisaacgymenvs.utils.task_util', 'omniisaacgymenvs.utils.config_utils', 'omniisaacgymenvs.utils.config_utils.path_utils', 'omniisaacgymenvs.envs', 'omniisaacgymenvs.envs.vec_env_rlgames', 'omni', 'omni.isaac.gym', 'omni.isaac.gym.vec_env', 'omni.isaac.gym.vec_env.vec_env_base', 'omni.kit.app._impl.telemetry_helpers', 'omni.isaac.gym.vec_env.vec_env_mt'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/apps/omni.isaac.sim.python.gym.headless.kit', '--/app/tokens/exe-path=/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/apps', '--/physics/cudaDevice=0', '--portable', '--no-window']
Passing the following args to the base kit application:  ['headless=True', 'task=Ant', 'multi_gpu=True']
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omniisaacgymenvs', 'omniisaacgymenvs.utils', 'omniisaacgymenvs.utils.hydra_cfg', 'omniisaacgymenvs.utils.hydra_cfg.hydra_utils', 'omniisaacgymenvs.utils.hydra_cfg.reformat', 'omniisaacgymenvs.utils.rlgames', 'omniisaacgymenvs.utils.rlgames.rlgames_utils', 'omniisaacgymenvs.utils.task_util', 'omniisaacgymenvs.utils.config_utils', 'omniisaacgymenvs.utils.config_utils.path_utils', 'omniisaacgymenvs.envs', 'omniisaacgymenvs.envs.vec_env_rlgames', 'omni', 'omni.isaac.gym', 'omni.isaac.gym.vec_env', 'omni.isaac.gym.vec_env.vec_env_base', 'omni.kit.app._impl.telemetry_helpers', 'omni.isaac.gym.vec_env.vec_env_mt'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/apps/omni.isaac.sim.python.gym.headless.kit', '--/app/tokens/exe-path=/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/apps', '--/physics/cudaDevice=0', '--portable', '--no-window']
Passing the following args to the base kit application:  ['headless=True', 'task=Ant', 'multi_gpu=True']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230504_201332.log
2023-05-05 06:13:32 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar/config'
2023-05-05 06:13:32 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar/config'
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230504_201332.log
2023-05-05 06:13:32 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.lidar/config'
2023-05-05 06:13:32 [1ms] [Warning] [omni.ext.plugin] [ext: omni.drivesim.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.0/exts/omni.drivesim.sensors.nv.radar/config'
[0.090s] [ext: omni.stats-0.0.0] startup
[0.090s] [ext: omni.stats-0.0.0] startup
[0.094s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.094s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.098s] [ext: omni.assets.plugins-0.0.0] startup
[0.099s] [ext: omni.gpu_foundation-0.0.0] startup
[0.097s] [ext: omni.assets.plugins-0.0.0] startup
[0.098s] [ext: omni.gpu_foundation-0.0.0] startup
2023-05-05 06:13:32 [96ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, because the interface was previously acquired by: 
2023-05-05 06:13:32 [95ms] [Warning] [carb] FrameworkImpl::setDefaultPlugin(client: omni.gpu_foundation_factory.plugin, desc : [carb::graphics::Graphics v2.11], plugin : carb.graphics-vulkan.plugin) failed. Plugin selection is locked, be [... cut ...]
========================================================
scripts/rlgames_train.py FAILED
--------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
--------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-05-04_20:13:55
  host      : dl
  rank      : 1 (local_rank: 1)
  exitcode  : -11 (pid: 321901)
  error_file: <N/A>
  traceback : Signal 11 (SIGSEGV) received by PID 321901
========================================================

kellyguo11, 2023-05-05:
Hi there, the error seems to suggest that one of the Isaac Sim instances was not launched on the correct GPU device. This was a new feature introduced in the latest Isaac Sim 2022.2.1 release, could you try updating to this version of Isaac Sim?

Robokan, 2023-05-05:
I updated from 2022.2.0 to 2022.2.1 and it does work! thanks.

closed 2023-05-05 by Robokan

Robokan, 2023-05-05:
only thing is it is using a very small amount of the memory and resources on each GPU. Any attempt to increase the number of environments just crashes with memory errors. AnymalTerrain uses 2048 and if I double it to 4096 it does not work. Even though I am only using a fraction of the memory on each GPU.