isaac-sim/OmniIsaacGymEnvs #46: Not possible to run on any GPU other than cuda:0

Robokan, 2023-05-05:
on line 106 of simulation_context.py

       if self._device is not None and "cuda" in self._device:
            device_id = self._settings.get_as_int("/physics/cudaDevice")
            self._device = f"cuda:{device_id}"
 
this will always return cuda:0 as far as I can tell. No way to pass in cuda:1 when constructing the context.

I have changed both device_id and rl_device in config.yaml in omniIsaacgymenvs and config.yaml in AnymalTerrain. So cuda: 1 seems to be used everywhere else but here.

kellyguo11, 2023-05-05:
Hi there, could you confirm the version of Isaac Sim you are working with? The GPU device should be set as a parameter when passed to the VecEnv class from rlgames_train.py. On the VecEnv side, this will be used when creating the Isaac Sim instance, which will set the `/physics/cudaDevice` value.

Robokan, 2023-05-05:
Thanks! Updating to 2022.2.1 doesn't crash when running the following command:

python scripts/rlgames_train.py task=AnymalTerrain checkpoint=runs/AnymalTerrain/nn/AnymalTerrain.pth test=True num_envs=64 rl_device=cuda:1 device_id=1

But it still shows cuda:0 at 97% capacity using 136 w and is allocating memory on it. If I stop the process cuda:0 goes down to 5%. So it's definitely causing it. It is using some resources on cuda:1 but why would it be using so much more on cuda:0?

|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4090         On | 00000000:01:00.0 Off |                  Off |
|  0%   37C    P2              136W / 450W|   6832MiB / 24564MiB |     97%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090         On | 00000000:41:00.0 Off |                  Off |
|  0%   33C    P2               84W / 450W|   4376MiB / 24564MiB |     61%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

python scripts/rlgames_train.py task=AnymalTerrain checkpoint=runs/AnymalTerrain/nn/AnymalTerrain.pth test=True num_envs=64 rl_device=cuda:1 device_id=1
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/bizon/anaconda3/envs/isaac-sim/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts', '--ext-folder', '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/apps', '--/physics/cudaDevice=1', '--portable']
Passing the following args to the base kit application:  ['task=AnymalTerrain', 'checkpoint=runs/AnymalTerrain/nn/AnymalTerrain.pth', 'test=True', 'num_envs=64', 'rl_device=cuda:1', 'device_id=1']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230505_080928.log
2023-05-05 18:09:28 [7ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar/config'
2023-05-05 18:09:28 [7ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar' or '/home/bizon/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar/config'
[0.105s] [ext: omni.stats-0.0.0] startup
[0.115s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.119s] [ext: omni.assets.plugins-0.0.0] startup
[0.120s] [ext: omni.gpu_foundation-0.0.0] startup
[0.125s] [ext: carb.windowing.plugins-1.0.0] startup
2023-05-05 18:09:28 [123ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  ASRock LED Controller [03000000ce260000a201000010010000]
[0.131s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 530.30.02     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 1 |     | 24810   MB | 10de      | 0          |
[... cut ...]
build mlp: 188
build mlp: 188
RunningMeanStd:  (1,)
RunningMeanStd:  (188,)
=> loading checkpoint '/home/bizon/eric/OmniIsaacGymEnvs/omniisaacgymenvs/runs/AnymalTerrain/nn/AnymalTerrain.pth'
[2023-05-05 08:09:45] Running RL reset
reward: 7.2 steps: 998.0
reward: 5.6 steps: 999.0
reward: 4.4 steps: 999.0
reward: 0.0 steps: 780.0
reward: 0.6 steps: 794.0
reward: 0.1 steps: 802.0
reward: 0.5 steps: 817.0
reward: 6.7 steps: 999.0
reward: 3.0 steps: 999.0
reward: 8.8 steps: 999.0
reward: 15.8 steps: 999.0