isaac-sim/OmniIsaacGymEnvs #47: When using Go1 USD model collisions do not work.

Robokan, 2023-05-07:
If I load it manually into isaac sim and press the simulation button it does collide with the ground. However if I use it instead of Anymal in the Anymal task (I have carefully set up all the joints correctly) It falls right though the ground plane during training.

Only thing I can think of is there is the warning:

[Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/envs/env_0/go1/RL_hip/collisions is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.

How do you set this setting?

kellyguo11, 2023-05-10:
Hi there, thanks for reporting the issue. We are aware of the collision issue with the Go1 asset and are investigating the cause. If you would like to workaround the issue, you can try adding the following snippet in the `post_reset()` method:

```
import omni.usd
stage = omni.usd.get_context().get_stage()
ground_prim = stage.GetPrimAtPath("/World/defaultGroundPlane/GroundPlane/CollisionPlane")
ground_prim.GetAttribute("physics:collisionEnabled").Set(False)
ground_prim.GetAttribute("physics:collisionEnabled").Set(True)
```

Robokan, 2023-05-12:
Great that works. Let me know when you get a fix. 

I have the research version of the Go1 and am working to transfer a NN to it via ROS. If you have links to any resources that would helpful it would be very much appreciated.

Robokan, 2023-05-14:
Do you have a fix for AnimalTerrain with the Go1? It also falls through all the terrain. Enabling and Disabling of the terrain doesn't seem to work.

djhanove, 2023-05-27:
@Robokan can you by chance share your go1.usd file? I also am trying to train the Go1 in the Anymal environment and having some issues

Robokan, 2023-06-02:
You can get it from the Nucleus server. Its under:
local host -> NVIDIA -> assets -> Isaac -> 2022.1.1 -> isaac -> robots -> unitree -> go1.usd

Zenif-NIght, 2023-06-07:
@djhanove did this work for you?
When I do this @Robokan  it seems the file size are the same 60MB
I wrote about this here: [Issue 55](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/55#issue-1744777479)

Robokan, 2023-06-26:
It does make 2 files the same size, but if you try it out as far as I can tell it uses less GPU. Have you tried to compare GPU usage?