isaac-sim/OmniIsaacGymEnvs #48: [Error] [carb.events.python] AttributeError: 'PlayButtonGroup' object has no attribute '_play_button'

johannespitz, 2023-05-09:
Hi,

when I set `enable_cameras=True` and try to run a training in `headless` mode (`$ isaacpython omniisaacgymenvs/scripts/rlgames_train.py task=AllegroHand num_envs=4 test=True headless=True`), I get this error 

```
[Error] [carb.events.python] AttributeError: 'PlayButtonGroup' object has no attribute '_play_button'
At:
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/exts/omni.kit.widget.toolbar/omni/kit/widget/toolbar/builtin_tools/play_button_group.py(129): _on_timeline_event
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(549): play
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(387): initialize_physics
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py(408): reset
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.core/omni/isaac/core/world/world.py(281): reset
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py(94): set_task
  /home/johannes/work/isaac-sim-envs/omniisaacgymenvs/envs/vec_env_rlgames.py(51): set_task
  /home/johannes/work/isaac-sim-envs/omniisaacgymenvs/utils/task_util.py(72): initialize_task
  omniisaacgymenvs/scripts/rlgames_train.py(114): parse_hydra_configs
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/core/utils.py(160): run_job
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/hydra.py(106): run
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py(381): <lambda>
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py(211): run_and_report
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/utils.py(378): _run_hydra
  /home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/main.py(52): decorated_main
  omniisaacgymenvs/scripts/rlgames_train.py(143): <module>
```

The simulation continues to run and using something like this we can even get camera images, but I would still like to resolve the error to make sure we don't run into problem later on.
```
       cam = rep.create.camera(
            position=position, look_at=look_at, clipping_range=(0.01, 10)
        )
        RESOLUTION = self._cfg["video_resolution"]
        self._render_product_path = rep.create.render_product(
            cam, resolution=(RESOLUTION, RESOLUTION)
        )
        self.rgb = rep.AnnotatorRegistry.get_annotator("rgb")
        self.rgb.attach([self._render_product_path])
        self.rgb.get_data()
```

kellyguo11, 2023-05-10:
Hi there, thanks for reporting the issue, we will have this fixed in the next release. The error should be harmless, but if you'd like to resolve it, you can try adding the following line `exts."omni.kit.widget.toolbar".PlayButton.enabled = false` in `apps/omni.isaac.sim.python.gym.headless.render.kit` under `[settings]`.

johannespitz, 2023-05-12:
Thanks, that works for now!

closed 2023-05-12 by johannespitz