isaac-sim/OmniIsaacGymEnvs #54: Is it possible to use WebSocket Browser Client

volkn, 2023-06-06:
Hello, I like to use WebSocket Browser Client rather than the Omniverse Streaming Client. Is this even possible? 

Thank you beforehand.

kellyguo11, 2023-06-21:
Hi there, it should be possible to use WebRTC and WebSocket Browser Clients by adding the following dependencies to the `omni.isaac.sim.python.gym.headless.render.kit` app file:

```
"omni.services.streamclient.websocket" = {}
"omni.services.streamclient.webrtc" = {}
```

Ramanathan0908, 2023-07-17:
Hi, after adding these dependencies how should I proceed? For example to run it in omniverse streaming client I just run `PYTHON_PATH scripts/rlgames_train.py task=Ant headless=True enable_livestream=True`.

maxencefaldor, 2024-01-19:
Hi,

Any update on this? It would be great to have WebRTC support for Isaac Gym!