isaac-sim/OmniIsaacGymEnvs #58: Go1.USD collisions in Anymal_Terrain do not seem to work properly

isaac-racine, 2023-06-20:
I am currently trying to implement the Go1.usd instead of the Anymal.usd in the Anymal_terrain task. At first the robot was falling right through the ground so I added the suggestion found in the [Issue 47](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/47) to disable and enable the collision of the the terrain :

`import omni.usd;`
`stage = omni.usd.get_context().get_stage()`
`ground_prim = stage.GetPrimAtPath("/World/defaultGroundPlane/GroundPlane/CollisionPlane")`
`ground_prim.GetAttribute("physics:collisionEnabled").Set(False)`
`ground_prim.GetAttribute("physics:collisionEnabled").Set(True)`

Now only the legs of the Go1 falls through the ground. The collision between the base and the ground works fine but I don't know why it doesn't work with the legs. 

I get no errors running the simulation headless or not. What's going on here? How to go about debugging the collision detection for the legs.

kellyguo11, 2023-06-21:
Hi there, thanks for reporting the issue. We are aware of some collision issues with the cylinder collision shapes on the Go1 and will have the assets updated for the next release.

isaac-racine, 2023-06-21:
Thank you for your reply! Is the date of the new release known or is it still far away?

Robokan, 2023-08-03:
If all cylinder collisions are removed from the Go1.urdf. It still falls through the terrain on animal terrain example. The only fix is to lower the number of environments to 40 or below and then it works. So I don't believe this is a cylinder collision problem but something deeper.