isaac-sim/OmniIsaacGymEnvs #65: Examples crash upon startup

LarsDoorenbos, 2023-07-27:
Hey,

I am trying to run the example `PYTHON_PATH scripts/rlgames_train.py task=Cartpole` given in the README. However, I run into the following error: ``Inconsistency detected by ld.so: ../elf/dl-tls.c: 618: _dl_allocate_tls_init: Assertion `listp != NULL' failed!.``

[Setting](https://stackoverflow.com/questions/22564780/debugging-ld-inconsistency-detected-by-ld-so) `LD_DEBUG=all` does not give more information. [This issue](https://github.com/ray-project/ray/issues/33858) in a different repo claims this is due to glibc bug, and it is fixed from 2.34. However, from `ldd version` I get `ldd (Ubuntu GLIBC 2.35-0ubuntu3.1) 2.35`, so that should not be the issue either.

Running Isaac Sim normally (from the Omniverse launcher or my own script) works as expected. I am running Ubuntu 22.04 with the built-in Python 3.7 environment of Isaac Sim.

 Here is the full log:

> /bin/bash: /home/ldoorenbos/anaconda3/lib/libtinfo.so.6: no version information available (required by /bin/bash)
/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1
/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:412: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.
> 
>   deprecation_warning(msg)
> [Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
> [Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
> Starting kit application with the following args:  ['/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts', '--ext-folder', '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/apps', '--/physics/cudaDevice=0', '--portable']
> Passing the following args to the base kit application:  ['task=Cartpole']
> [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
> [Info] [carb] Logging to file: /home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230727_172128.log
> 2023-07-27 15:21:28 [30ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar' or '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.lidar/config'
> 2023-07-27 15:21:28 [30ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar' or '/home/ldoorenbos/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.sensors.nv.radar/config'
> [0.219s] [ext: omni.stats-0.0.0] startup
> [0.237s] [ext: omni.rtx.shadercache-1.0.0] startup
> [0.245s] [ext: omni.assets.plugins-0.0.0] startup
> [0.247s] [ext: omni.gpu_foundation-0.0.0] startup
> [0.254s] [ext: carb.windowing.plugins-1.0.0] startup
> [0.282s] [ext: omni.kit.renderer.init-0.0.0] startup
> 
> |---------------------------------------------------------------------------------------------|
> | Driver Version: 535.86.05     | Graphics API: Vulkan
> |=============================================================================================|
> | GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
> |     |                                  |        |     |            | Device-ID | UUID       |
> |---------------------------------------------------------------------------------------------|
> | 0   | NVIDIA RTX A6000                 | Yes: 0 |     | 49386   MB | 10de      | 0          |
> |     |                                  |        |     |            | 2230      | 6ccf01aa.. |
> |=============================================================================================|
> | OS: Linux f9409cc-lcedt, Version: 5.19.0-50-generic
> | XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
> | Processor: Intel(R) Core(TM) i9-10980XE CPU @ 3.00GHz | Cores: Unknown | Logical: 36
> |---------------------------------------------------------------------------------------------|
> | Total Memory (MB): 31795 | Free Memory: 23121
> | Total Page/Swap (MB): 15359 | Free Page/Swap: 15359
> |---------------------------------------------------------------------------------------------|
> [1.138s] [ext: omni.kit.pipapi-0.0.0] startup
> [1.147s] [ext: omni.kit.pip_archive-0.0.0] startup
> [1.149s] [ext: omni.kit.loop-isaac-1.0.0] startup
> [1.150s] [ext: omni.kit.async_engine-0.0.0] startup
> [1.151s] [ext: omni.kit.test-0.0.0] startup
> [1.170s] [ext: omni.usd.config-1.0.0] startup
> [1.174s] [ext: omni.usd.libs-1.0.0] startup
> [1.371s] [ext: omni.isaac.core_archive-2.0.1] startup
> [1.436s] [ext: omni.pip.torch-1_13_1-0.1.4] startup
> [1.438s] [ext: omni.isaac.ml_archive-1.1.0] startup
> [1.439s] [ext: omni.client-0.1.1] startup
> [1.448s] [ext: omni.appwindow-1.0.1] startup
> [1.456s] [ext: omni.kit.renderer.core-0.0.0] startup
> [1.565s] [ext: omni.kit.renderer.capture-0.0.0] startup
> [1.569s] [ext: omni.kit.renderer.imgui-0.0.0] startup
> [1.656s] [ext: carb.audio-0.1.0] startup
> [1.663s] [ext: omni.ui-2.14.4] startup
> [1.686s] [ext: omni.uiaudio-1.0.0] startup
> [1.688s] [ext: omni.kit.mainwindow-1.0.0] startup
> [1.689s] [ext: omni.kit.uiapp-0.0.0] startup
[... 207 lines cut ...]
> [4.203s] [ext: omni.kit.window.console-0.2.3] startup
> [4.214s] [ext: omni.physx.bundle-104.2.4-5.1] startup
> [4.214s] [ext: omni.graph.bundle.action-1.3.0] startup
> [4.214s] [ext: omni.kit.window.script_editor-1.7.2] startup
> [4.216s] [ext: omni.kit.hotkeys.window-1.1.3] startup
> [4.222s] [ext: omni.kit.window.status_bar-0.1.5] startup
> [4.227s] [ext: omni.kit.window.title-1.1.2] startup
> [4.228s] [ext: omni.replicator.composer-1.2.10] startup
> [4.233s] [ext: omni.replicator.isaac-1.7.4] startup
> [4.247s] [ext: omni.resourcemonitor-1.0.0] startup
> [4.254s] [ext: omni.rtx.settings.core-0.5.8] startup
> [4.259s] [ext: omni.isaac.franka-0.4.0] startup
> [4.260s] [ext: omni.kit.viewport.rtx-104.0.0] startup
> [4.260s] [ext: semantics.schema.editor-0.3.3] startup
> [4.263s] [ext: omni.kit.widget.live-2.0.3] startup
> [4.266s] [ext: omni.isaac.utils-0.2.4] startup
> [4.270s] [ext: omni.isaac.kit-1.4.1] startup
> [4.270s] [ext: omni.isaac.cortex-0.3.2] startup
> [4.271s] [ext: omni.kit.window.stats-0.1.2] startup
> [4.272s] [ext: omni.isaac.sim.python-2022.2.1] startup
> [4.273s] Simulation App Starting
> Inconsistency detected by ld.so: ../elf/dl-tls.c: 618: _dl_allocate_tls_init: Assertion `listp != NULL' failed!
> There was an error running python

Any help/ideas would be much appreciated.

sambaPython24, 2023-08-30:
I encountered the same problem on Ubuntu. Did you make any progress with that problem?

LarsDoorenbos, 2023-08-31:
No, unfortunately I have not been able to solve this.