isaac-sim/OmniIsaacGymEnvs #66: Adding objects post-cloning

lenroe, 2023-08-02:
Adding prims to the stage after calling ```omni.isaac.cloner.GridCloner.clone()``` results in not expected behavior. 
When adding a prim to the ```source_prim_path``` _after_  the cloning process was already triggered,  the prim will be duplicated to all environments. 

Below is a minimal example illustrating the issue.

```
from omni.isaac.cloner import GridCloner
from omni.isaac.core.utils.stage import print_stage_prim_paths
from omni.isaac.core.objects import DynamicSphere

num_envs = 3
base_env_path = '/World/envs'
zero_env_path = '/World/envs/env_0'

# adding a sphere to env_0 before calling the cloner
ball = DynamicSphere(prim_path= zero_env_path +'/sphere_0')

cloner = GridCloner(spacing=0.1)
prim_paths = cloner.generate_paths('/World/envs/env', num_envs)
cloner.clone(source_prim_path=zero_env_path, prim_paths=prim_paths, base_env_path=base_env_path)

print_stage_prim_paths() # as expected, all envs have a "sphere_0" prim

ball = DynamicSphere(prim_path= f"{base_env_path}/env_1/sphere_1")

print_stage_prim_paths() # 'sphere_1' prim was added to env_1 (as expected)

ball = DynamicSphere(prim_path= f"{base_env_path}/env_0/sphere_2")

print_stage_prim_paths() # 'sphere_2' prim was added to env_0, but also to all other envs (NOT expected)
```
The output of the last ```print_stage_prim_paths()``` statement is:
...
/World/envs
/World/envs/env_0
/World/envs/env_0/sphere_0
/World/envs/env_0/sphere_2
/World/envs/env_1
/World/envs/env_1/sphere_0
/World/envs/env_1/sphere_2
/World/envs/env_1/sphere_1
/World/envs/env_2
/World/envs/env_2/sphere_0
/World/envs/env_2/sphere_2
...

When the expected result would be:

...
/World/envs
/World/envs/env_0
/World/envs/env_0/sphere_0
/World/envs/env_0/sphere_2
/World/envs/env_1
/World/envs/env_1/sphere_0
/World/envs/env_1/sphere_1
/World/envs/env_2
/World/envs/env_2/sphere_0
...

I believe this also might cause a bug in [factory_env_nut_bolt.py](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/tasks/factory/factory_env_nut_bolt.py#L116), where only one `subassembly` (i.e. the one that is added to ```env_0```) is duplicated to all environments.

If this behavior is intentional, what would be the recommended way of adding different prims to each ```env``` when using the ```clone()``` method?

kellyguo11, 2023-08-02:
Hi there, yes this behaviour is expected with the current cloner for performance reasons. We will be introducing a new flag to the cloner for the next release that will allow for a different copying behaviour, which will not propagate changes made in the source environment to the cloned environments.

lenroe, 2023-08-03:
Great, thanks!

closed 2023-08-03 by lenroe