isaac-sim/OmniIsaacGymEnvs #67: Change parameters of the physics scene

dhruvkm2402, 2023-08-03:
Hi,
I want to change parameters of the physics scene like disabling GPU dynamics, enabling CCD, change the solver to PGS and boradphase type to MBP before launching the training script as it works best for a skid-steered platform. What is the best way to do it?

kellyguo11, 2023-08-28:
Hi there, you can achieve this by using the `physics_context` object in Isaac Sim: https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#module-omni.isaac.core.physics_context. This can be accessed from the task environment through the world object.

dhruvkm2402, 2023-09-25:
Hi @kellyguo11 
When I import PhysicsContext and then do
PhysicsContext().enable_gpu_dynamics(flag=False) in the set_up_scene function following is the error I get:
return pose[indices, 0:3], pose[indices, 3:7][:, [3, 0, 1, 2]]
RuntimeError: indices should be either on cpu or on the same device as the indexed tensor (cpu)
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7fcc793a8e60>

What could be the issue?