isaac-sim/OmniIsaacGymEnvs #71: Add to documentation: How to load custom objects to set_up scene

sambaPython24, 2023-08-25:
Could you add to the documentation (or give a python example in ```task\ .... .py```) 
how custom objects and robots are loaded from a local file. 

I saw, that objects in the examples are loaded primarily from ```omni.isaac.core``` (such as ```omni.isaac.core.robots.robot``` or ```omni.isaac.core.object``` for which the ```assets root_path``` is given by ```http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets``` but not accessible. 

It would probably not just help me to have a simple example, in which files from a local file a loaded into a scene and then used for RL learning.

kellyguo11, 2023-08-28:
Hi there, `http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets` is the Nucleus server where we host all of the Isaac Sim assets, it should be accessible within Isaac Sim or Nucleus. To add an object from a local path, you can simply provide the file path to the following API:

`add_reference_to_stage(file_path, prim_path)`

sambaPython24, 2023-08-29:
Thanks for your comment but the next question arises, how to create a simple, custom object to pass to ```world.scene.add```.
In the [Hello World](https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_core_hello_robot.html) example, the predefined robot class is used. 

While that might be just a finger excercise for you, seing that once with very simple assets would be great to have in the documentation.