isaac-sim/OmniIsaacGymEnvs #72: Factory (Fast Contact for Robotic Assembly) integration

pt127, 2023-08-31:
Hello,

is there a  a plan when and if  "Factory" will be integrated in OmniIsaacGym as already for the Isaac Gym - Preview Release?
Is there a way to already use the controllers provided in Factory in the OmniIsaacGym envs?

Thanks

iakinola23, 2023-10-09:
Hi,
Factory is already integrated into OmniIsaacGym, along with the associated robot [controllers](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/tasks/factory/factory_control.py). The Pick task from Factory is already available [here](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/tasks/factory/factory_task_nut_bolt_pick.py), and we anticipate releasing the remaining Factory tasks in the coming weeks.
We hope you'll find the Pick task and controllers already useful for your applications.
Best,

pt127, 2023-10-09:
thanks for the answer. When i change the controller in the FactoryTaskNutBoltPick.yaml from joint_space_id  to task_space_impedance, there iss something wrong with the robot movement. The wrist is spinning around randonly. Is there a fix for that? Thanks

iakinola23, 2023-10-10:
Hi,

I suspect that the issue may be due to the default gains for the task_space_impedance controller not being tuned for the robot model. I suggest adjusting the gains, particularly focusing on the rotational P/D gains. As a starting point, you can consider reducing the D gains by an order of magnitude.

Please give this a try and let us know if it improves the situation.

Best,

closed 2023-11-28 by kellyguo11