isaac-sim/OmniIsaacGymEnvs #74: Aliengo penetrating through the ground on the AnymalTerrain example

shafeef901, 2023-09-06:
I've tried loading the aliengo URDF ([aliengo_description](https://github.com/unitreerobotics/unitree_ros/tree/master/robots/aliengo_description/xacro)) on the AnymalTerrain example (anymal_terrain.py) but the feet are penetrating through the ground as shown in the video below:

https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/assets/24255993/5d315d47-813b-4b89-a27b-16cdec546488

To import the URDF, I updated the __init__() of Anymal class in robots/articulations/anymal.py with the following code:

```
status, import_config = omni.kit.commands.execute("URDFCreateImportConfig")
import_config.merge_fixed_joints = False
import_config.convex_decomp = True
import_config.import_inertia_tensor = True
import_config.fix_base = False
import_config.self_collision = False

# import URDF
success, robot_prim_path_default = omni.kit.commands.execute(
    "URDFParseAndImportFile",
    urdf_path=urdf_path,
    import_config=import_config, 
)

move_prim(robot_prim_path_default, prim_path)
```

How do I fix this?

sujitvasanth, 2023-11-04:
rather than creating the usd file on the go suggest creating a usd asset using urdf omporter in IsaacSIM and inspecting the resultant usd in Isaac Sim and then loading that usd rather than the urdf.

In the stage view check the foot and tibia(or equivalent)  xforms have rigid body attributes and if not add them
  in your video you can see the feet have some contact properties as some feet do stay above the ground plane

The other issue I came across you should look at the task/*.yaml to check the initialization height of your robot base is appropriate for the distance between the ground plane. You can debug this by setting a very high initiation height

lastly look for any warnings about the size of your gpu physics buffers which I again found problems - this again can be specified in the task's yaml file.

In general, you can tell if the problem is with IsaacSim itself or the usd by:
opening a single instance of the usd in IsaacSim - you can add a physics ground plane and run the simulation and move the joints angles in the inspector .. If the robot is behaving appropriately then the problem is in the omniverseenvs implementation rules ie physics resets, and base initialization. gpu memory parameters

It took 2-3 weeks of frustration and trial and error to get this to work for me!
There seems to have been some implementation issues of usd robots in 2022.2.1 which led to esoteric errors which are partly resolved in 2023.1.0 so suggest an upgrade if you have not done so..(you can have both versions installed simultaneously) note that you will have to upgrade you omniisaacgymenvs repo to maintain compatibility.  Suggest keeping a backup and also having fresh install folders for the new versions of IsaacSim and omniisaacgymenvs

mohamedamr13, 2024-01-16:
@sujitvasanth I am currently running on Isaac Sim 2023.1.1 but I also have an issue running the anymal terrain example using the go1.usd. I have documented the issue [here](https://forums.developer.nvidia.com/t/anymsalterrain-training-failed-with-unitree-go1-instead-of-anybiotics-anymal/278892). Do you have any possible solution for this ?