isaac-sim/OmniIsaacGymEnvs #76: Implementation guides

Pipe-Runner, 2023-09-20:
I know there are better places to ask than this, but is there a recommended resource list that talks and guides through the implementation details of an RL example from scratch? The examples provided need to be simplified.
Unless I am missing pre-requisites, the FrankaCabinet example code is too hard to understand. The `framework.md` does not do justice to the complexity of the tool. Sorry if this is a silly question; feel free to close this ticket.

sambaPython24, 2023-09-22:
You are right and the missing resources for learning how to create your own examples is a major source of frustration in the community. Everybody is just trying-and-erroring through various sources of documentation, videos and tutorials while stumbling over bugs. 

There is a list of examples for [Isaac Gym](https://github.com/wangcongrobot/awesome-isaac-gym) (which is the older version of this repo).

One could try to adapt the FrankaCabinet scene.

Pipe-Runner, 2023-09-28:
@sambaPython24 Thanks for your input. I am currently just trying to retrofit the example in this repo but it's hard to reason with a lot of things in an example with such complexity. I currently have a few videos open in my tap which go over the old repo (https://www.youtube.com/watch?v=QrlUznrbW8I) I don't expect much but maybe the person in the video will go over the terminologies at least.

Toni-SM, 2023-09-28:
Hi @Pipe-Runner 

I encourage you to take a look at the [skrl's real world example code for the Franka Panda](https://skrl.readthedocs.io/en/latest/intro/examples.html#real-world-examples), in particular the simulation in Omniverse Isaac Gym Environments (OIGE). This example (very simple, to allow later modifications) is implemented using the Omniverse Isaac Gym framework (as described in `framework.md`), the main difference is that the implementation is done in a file decoupled from the OIGE file structure (it is not necessary to modify the latter to integrate the task).

Pipe-Runner, 2023-09-28:
> Hi @Pipe-Runner
>
> I encourage you to take a look at the [skrl's real world example code for the Franka Panda](https://skrl.readthedocs.io/en/latest/intro/examples.html#real-world-examples), in [...]

As it turns out, I actually have been referring to the exact code you are talking about. They also implement the task using the RLTask base class which is good since it becomes a bit easy to follow. The example that they have is much easier to understand but drawing parallel to the code in OIGE for similar tasks is quite confusing especially with a lot of utility code for coordinate system conversation.
After spending quite some time, most of the structure has started making sense but the nuances are still fuzzy. Do you know if there is a place where I can ask questions or maybe just talk to people who are using OIGE or derived libs?

closed 2023-11-04 by Pipe-Runner