isaac-sim/OmniIsaacGymEnvs #80: Infer trained neural network manually

BaranAugustus, 2023-10-04:
Hello!

So I have trained 3 neural networks using Omni Isaac Gym Envs and the rl_games RL library. Now I want to define a new task. In this new task I want to infer from the already trained neural networks. So in the `pre_physics_step` I want to be able to say
`If condition_0: actions = model_0(obs), else if condition_1: actions=model_1(obs)` . I know you can run inference by setting test=True and specifying a corresponding checkpoint, but I want to use multiple networks manually in the code, in my case an OmniIsaacGym Task file.

Think of **Hierarchical Reinforcement Learning** where you have a higher level policy commanding a lower level policy. When training the higher level policy you need to be able to infer from the already trained lower level policy. Perhaps take a look at the code above the sentence "We should see our cartpole policy running!" in this link: https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_gym_new_rl_example.html. I, too, want to be able to write  `action = model.predict(obs)`.

I have taken this issue into account: https://github.com/Denys88/rl_games/issues/243
Since my knowledge of torch/neural networks is still limited, I do not know if the solution to the linked issue is also the solution to my issue. I expected an easier way to incorporate an already trained neural network into my task file.

sujitvasanth, 2023-11-06:
Hi Baran.. i think most pytorch functions will be run on GPU because this is how pytorch is designed but the issue will be that you will need to recreate the inputs you set for your trained neural network. Its likely this might be unnecessarily complex and you might need to look into this to make it work efficiently by removing uneccessary inputs and layers from the NN training that you thought you needed but actually didn't and thus making the NN's as small a possible.
I can't see any reason you can't run inference of another neural network on the GPU as part of your actor actuation while simultaneously training another one. just will be more complex than just setting a joint angle etc..If you compartmentalize the actuation - it doesn't actually matter what method you calculate the actuator positions i.e neural network, function, etc
you would load the NN's onto the GPU so can be reused throughout the training
I read denys88 response https://github.com/Denys88/rl_games/issues/243 which says basically the same

eferreirafilho, 2024-03-18:
Hi @BaranAugustus, I'm facing a similar crossroads, have you managed to solve that?
Many thanks!