isaac-sim/OmniIsaacGymEnvs #85: Isaac Sim 2023.1 CartpoleCamera Task not loading

dhruvkm2402, 2023-10-18:
Hi @kellyguo11 
Parallelization through the camera is a great new feature. While testing the cartpole in Isaac Sim 2023.1, following is the warning that comes up and the app never launches. I tried waiting for a couple of minutes

PYTHON_PATH scripts/rlgames_train.py task=CartpoleCamera
/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(
Starting kit application with the following args:  ['/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/dhruvm/OmniIsaacGymEnvs/apps/omni.isaac.sim.python.gym.camera.kit', '--/app/tokens/exe-path=/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts', '--ext-folder', '/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  ['task=CartpoleCamera']
[Info] [carb] Logging to file: /home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/logs/Kit/omni.isaac.sim.python.gym.camera/2023.1/kit_20231018_172016.log
2023-10-18 21:20:16 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.040s] Failed to solve some dependencies locally, syncing with extension registry...
[0.042s] [ext: omni.kit.async_engine-0.0.0] startup
[0.043s] [ext: omni.assets.plugins-0.0.0] startup
[0.045s] [ext: omni.client-1.0.1] startup
[0.052s] [ext: omni.kit.registry.nucleus-0.0.0] startup

Is there any additional setting that needs to be setup?

dhruvkm2402, 2023-10-18:
And following is the error:

2023-10-18 21:33:00 [131,368ms] [Error] [omni.kit.registry.nucleus.utils.common] syncing index failed. URL: 'omniverse://kit-extensions.ov.nvidia.com/exts/kit/default/3'. Result: Result.ERROR_CONNECTION
2023-10-18 21:33:00 [131,368ms] [Error] [omni.ext.plugin] Syncing extension registry failed: registry dict is nullptr.
2023-10-18 21:33:00 [131,369ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[omni.isaac.sim.python.gym.camera-2023.1.0 -> omni.isaac.sim.python.gym-2023.1.0] dependency: (name: 'omni.kit.menu.stage', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
	 (none found)

2023-10-18 21:33:00 [131,369ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
[131.395s] Simulation App Starting
2023-10-18 21:33:00 [131,371ms] [Error] [omni.kit.app._impl] [py stderr]: Error executing job with overrides: ['task=CartpoleCamera']

Error executing job with overrides: ['task=CartpoleCamera']
2023-10-18 21:33:00 [131,459ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):

Traceback (most recent call last):
2023-10-18 21:33:00 [131,459ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,459ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/dhruvm/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 98, in parse_hydra_configs
    env = VecEnvRLGames(

  File "/home/dhruvm/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 98, in parse_hydra_configs
    env = VecEnvRLGames(
2023-10-18 21:33:00 [131,459ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,459ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py", line 56, in __init__
    self._simulation_app = SimulationApp(

  File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py", line 56, in __init__
    self._simulation_app = SimulationApp(
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 214, in __init__
    create_new_stage()

  File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 214, in __init__
    create_new_stage()
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 65, in create_new_stage
    import omni.usd

  File "/home/dhruvm/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.kit/omni/isaac/kit/utils.py", line 65, in create_new_stage
    import omni.usd
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: ModuleNotFoundError: No module named 'omni.usd'

ModuleNotFoundError: No module named 'omni.usd'
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: 
2023-10-18 21:33:00 [131,460ms] [Error] [omni.kit.app._impl] [py stderr]: 
Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2023-10-18 21:33:00 [131,463ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
There was an error running python

kellyguo11, 2023-10-19:
Hi there, thanks for reporting the issue. Could you please try replacing the last line in apps/omni.isaac.sim.python.gym.camera.kit with `folders = ["${kit}/../exts", "${kit}/../extscache", "${kit}/../extsPhysics", "${kit}/exts", "${kit}/extscore"]`

dhruvkm2402, 2023-10-20:
Hello, I didn't find any file with the given name. There's only omni.isaac.sim.python.gym
I tried replacing the last line in it but still the same error.

robinvetsch, 2023-10-20:
> Hi there, thanks for reporting the issue. Could you please try replacing the last line in apps/omni.isaac.sim.python.gym.camera.kit with `folders = ["${kit}/../exts", "${kit}/../extscache", "${kit}/../extsPhysics", "${kit}/exts", "${kit}/extscore"]`

I had the same issue. Replacing the last line in the file worked for me. Many Thanks

kellyguo11, 2023-10-20:
Thanks for confirming. I've added the fix to the repo.

closed 2023-10-20 by kellyguo11

Guanbin-Huang, 2024-05-01:
i already replaced the last line with what you said. but it sill doesn't load anything.

[image]

[image]

Guanbin-Huang, 2024-05-01:
i'm using isaac sim 2023.1.1

Guanbin-Huang, 2024-05-01:
it works!!!