isaac-sim/OmniIsaacGymEnvs #86: Cannot launch Isaac Sim for extension workflow

Virlus, 2023-10-19:
I encountered the following error while launching isaac sim for extension workflow, which causes failure to open the GUI for Isaac Sim:
```
2023-10-19 13:05:04 [3ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad-200.0.0-rc.4+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [3ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad_core-200.0.0-rc.3+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [3ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.core-103.4.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [3ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.usd-103.4.2+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [4ms] [Warning] [omni.ext.plugin] [ext: omni.kit.widget.timeline-105.0.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [4ms] [Warning] [omni.ext.plugin] [ext: omni.kit.window.sequencer-103.4.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [4ms] [Warning] [omni.ext.plugin] [ext: omni.paint.brush.attributes-1.3.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-19 13:05:04 [4ms] [Warning] [omni.ext.plugin] [ext: omni.usd.schema.sequence-2.3.0+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
[0.060s] Failed to solve some dependencies locally, syncing with extension registry...
[0.062s] [ext: omni.kit.async_engine-0.0.0] startup
[0.063s] [ext: omni.assets.plugins-0.0.0] startup
[0.063s] [ext: omni.client-1.0.1] startup
[0.067s] [ext: omni.kit.registry.nucleus-0.0.0] startup
```
The shell script that I've written is demonstrated below:
```
#!/bin/bash
/home/wanghanqing/ov/pkg/isaac_sim-2023.1.0/isaac-sim.gym.sh --ext-path /home/dev/wenye/OmniIsaacGymEnvs/config/extension.toml
```
The error emerges as I execute the script above. What could be the cause and how to fix this?

Virlus, 2023-10-19:
Also, the process would be shut down after a while automatically and there comes further error message as below:
```
2023-10-19 12:55:24 [130,821ms] [Error] [omni.kit.registry.nucleus.utils.common] syncing index failed. URL: 'omniverse://kit-extensions.ov.nvidia.com/exts/kit/default/3'. Result: Result.ERROR_CONNECTION
2023-10-19 12:55:24 [130,821ms] [Error] [omni.ext.plugin] Syncing extension registry failed: registry dict is nullptr.
2023-10-19 12:55:24 [130,821ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
        [omni.isaac.sim.gym-2023.1.0] dependency: (name: 'omniisaacgymenvs', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
         (none found)

2023-10-19 12:55:24 [130,821ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
[130.827s] [ext: omni.kit.registry.nucleus-0.0.0] shutdown
[130.828s] [ext: omni.client-1.0.1] shutdown
[130.829s] [ext: omni.assets.plugins-0.0.0] shutdown
[130.829s] [ext: omni.kit.async_engine-0.0.0] shutdown
2023-10-19 12:55:24 [130,839ms] [Warning] [carb] [Plugin: carb.dictionary.plugin] Module /home/wanghanqing/ov/pkg/isaac_sim-2023.1.0/kit/kernel/plugins/libcarb.dictionary.plugin.so remained loaded after unload request
```

kellyguo11, 2023-10-19:
Hi there, could you please try the command [here](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs#extension-workflow), using the `--ext-folder` argument.

Virlus, 2023-10-19:
@kellyguo11 Hi, the error message remains the same with the `--ext-folder` argument.

kellyguo11, 2023-10-19:
Can you share the full command you are using? Just to confirm, the `--ext-folder` flag should point to the parent directory of the OmniIsaacGymEnvs repo, and not the extension.toml file.

Virlus, 2023-10-19:
```
#!/bin/bash
/home/wanghanqing/ov/pkg/isaac_sim-2023.1.0/isaac-sim.gym.sh --ext-folder /home/dev/wenye
```

kellyguo11, 2023-10-19:
Thanks for confirming, we'll probably have to look into this. Can you try replacing the content of `apps/omni.isaac.sim.gym.kit` in the isaac_sim folder  with the below and see if it helps.

```
[package]
title = "Isaac Sim Python - Gym"
description = "A simplifed app for running Gym examples with extension workflow"
version = "2023.1.0"

# That makes it browsable in UI with "experience" filter
keywords = ["experience", "app", "usd"]

[dependencies]
"omni.isaac.sim.python.gym.headless" = {}

"omni.kit.material.library" = {}
"omni.kit.viewport.rtx" = {}
"omni.kit.viewport.bundle" = {}

"omni.kit.window.file" = {}
"omni.kit.window.script_editor" = {}
"omni.kit.window.status_bar" = {}
"omni.kit.window.title" = {}
"omni.kit.window.extensions" = {}
"omni.kit.window.toolbar" = {}

[settings]
exts."omni.kit.widget.toolbar".PlayButton.enabled = true

# set the default ros bridge to disable on startup
isaac.startup.ros_bridge_extension = ""

[settings.exts."omni.kit.registry.nucleus"]
registries = [
	{ name = "kit/default", url = "https://ovextensionsprod.blob.core.windows.net/exts/kit/prod/shared" },
	{ name = "kit/sdk", url = "https://ovextensionsprod.blob.core.windows.net/exts/kit/prod/sdk/${kit_version_short}/${kit_git_hash}" },
	{ name = "kit/community", url = "https://dw290v42wisod.cloudfront.net/exts/kit/community" },
]

# Register extension folder from this repo in kit
[settings.app.exts]
folders = ["${app}/../exts", "${app}/../extscache", "${app}/../extsPhysics", "${app}/../kit/exts"]

# Isaac Sim Extensions
###############################
[dependencies]
"omni.isaac.ui" = {}
"omni.isaac.app.setup" = { order = 1000 } # we are running that at the end
"omniisaacgymenvs" = { order = 999 }
```

Virlus, 2023-10-21:
@kellyguo11 The error message still pops up with the content replaced. I will present the entire error message below for your convenience:
```
2023-10-21 02:30:00 [6ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad-200.0.0-rc.4+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [6ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad_core-200.0.0-rc.3+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [6ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.usd-103.4.2+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [6ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.core-103.4.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [7ms] [Warning] [omni.ext.plugin] [ext: omni.kit.widget.timeline-105.0.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [7ms] [Warning] [omni.ext.plugin] [ext: omni.kit.window.sequencer-103.4.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [7ms] [Warning] [omni.ext.plugin] [ext: omni.paint.brush.attributes-1.3.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-10-21 02:30:00 [13ms] [Warning] [omni.ext.plugin] [ext: omni.usd.schema.sequence-2.3.0+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
[0.076s] Failed to solve some dependencies locally, syncing with extension registry...
[0.078s] [ext: omni.kit.async_engine-0.0.0] startup
[0.078s] [ext: omni.assets.plugins-0.0.0] startup
[0.079s] [ext: omni.client-1.0.1] startup
[0.083s] [ext: omni.kit.registry.nucleus-0.0.0] startup
2023-10-21 02:30:15 [14,746ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
        [omni.isaac.sim.gym-2023.1.0] dependency: (name: 'omniisaacgymenvs', tag: '', version: '', exact: 0, optional: 0) can't be satisfied. Available versions:
         (none found)

2023-10-21 02:30:15 [14,746ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...
[14.762s] [ext: omni.kit.registry.nucleus-0.0.0] shutdown
[14.766s] [ext: omni.client-1.0.1] shutdown
[14.768s] [ext: omni.assets.plugins-0.0.0] shutdown
[14.769s] [ext: omni.kit.async_engine-0.0.0] shutdown
2023-10-21 02:30:15 [14,770ms] [Warning] [carb] [Plugin: carb.dictionary.plugin] Module /home/wanghanqing/ov/pkg/isaac_sim-2023.1.0/kit/kernel/plugins/libcarb.dictionary.plugin.so remained loaded after unload request
```

kellyguo11, 2023-10-21:
Thanks, I think I know what the issue is. The github repo name is case sensitive, so it can't find the extension name. To fix it, you can change the last line of `apps/omni.isaac.sim.gym.kit` (in isaac sim), replacing `"omniisaacgymenvs" = { order = 999 }` with `"OmniIsaacGymEnvs" = { order = 999 }`, or rename the repo folder from `OmniIsaacGymEnvs` to `omniisaacgymenvs` locally. Let me know if either of those approaches work for you.

Virlus, 2023-10-22:
@kellyguo11 Hi there. I'm glad that both of these approaches worked so that I'm able to launch the isaac sim app now. However, the `Isaac Examples` button is missing from the menu, and here's what my isaac sim app looks like:
[image]

kellyguo11, 2023-10-22:
Likely an error occurred while it was trying to load the extension. You can try uncommenting the last two lines here: https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/__init__.py#L33 and removing `pass` to get a more useful error message. Please also make sure to run `pip install -e .` before launching the app.

Virlus, 2023-10-22:
I got the following error message, but by running `PYTHON_PATH -m pip list` I confirmed that I already had `hydra-core` installed. What might be the problem? @kellyguo11 
```
No module named 'hydra'
Traceback (most recent call last):
  File "/home/dev/wenye/OmniIsaacGymEnvs/omniisaacgymenvs/__init__.py", line 31, in <module>
    from .extension import RLExtension, get_instance
  File "/home/dev/wenye/OmniIsaacGymEnvs/omniisaacgymenvs/extension.py", line 17, in <module>
    import hydra
ModuleNotFoundError: No module named 'hydra'
```

kellyguo11, 2023-10-23:
Interesting, are you able to run any of the examples by launching from python directly (i.e. with `PYTHON_PATH scripts/rlgames_train.py task=TASK`)? The two workflows should be using the same python environment. Otherwise, maybe try just doing the pip install one more time to make sure all of the packages were installed without issues.

Pipe-Runner, 2023-10-26:
Same on my end:
[image]

I have also verified, I am able to run the training task. Also, I can also find `hydra-core` in my pip list. I am not using the python Isaac provides but using a conda env.

kellyguo11, 2023-10-27:
Thanks for confirming. Are you also seeing the same error that hydra is missing? We have not tried this with other python environments outside of Isaac yet so we'll have to look into this.

kellyguo11, 2023-11-06:
Glad to hear it works with the built-in python, one thing you can try with the conda environment is to force install the packages with `pip install -e . --force-reinstall`. The stage window has been hidden for performance reasons, it can be enabled from the Extension Manager by searching for `omni.kit.window.stage`.

limshoonkit, 2023-12-03:
Isaac sim version: isaac_sim-2023.1.0-hotfix.1
OmniIsaacGymEnvs: master branch

I am also not seeing any Isaac Example in the top menu.
[image]

No dependency issue on my end. Extension loaded just fine.
[image]

RL application can work fine through command line.

leoliumessi, 2024-01-03:
I have no the choice of "extension" in my window menu T.T
[image]

dantes-ai, 2024-01-24:
I'm facing the same issue as OP. I ran OmniIsaacGymEnv successfully after initial install.
I then moved the OmniIsaacGymEnv directory to another location on disk, and running Gym failed ever since (I updated the ext-folder accordingly...)

force-reinstall or any other method did not help....

eferreirafilho, 2024-03-21:
Same problem here, Isaac launches, but the RL extension does not appear. Tried in two different machines (RTX 3080 and RTX A5000). 

Tried in both
Isaac Sim 2023.1.1
and
Isaac Sim 2023.0.1-hotfix.1

All the steps I used:
```
mkdir ~/omniverse-playground
cd ~/omniverse-playground
git clone https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs
alias PYTHON_PATH=~/.local/share/ov/pkg/isaac_sim-2023.1.1/python.sh
cd ~/omniverse-playground/OmniIsaacGymEnvs
PYTHON_PATH -m pip install -e .
cd omniisaacgymenvs
~/.local/share/ov/pkg/isaac_sim-2023.1.1/isaac-sim.gym.sh --ext-folder ~/omniverse-playground
```

Terminal log:
```
nfdev@ws1:~/omniverse-playground$ ~/.local/share/ov/pkg/isaac_sim-2023.1.1/isaac-sim.gym.sh --ext-folder ~/omniverse-playground
Loading user config located at: '/home/nfdev/.local/share/ov/data/Kit/omni.isaac.sim.gym/2023.1/user.config.json'
[Info] [carb] Logging to file: /home/nfdev/.nvidia-omniverse/logs/Kit/omni.isaac.sim.gym/2023.1/kit_20240321_191834.log
2024-03-21 18:18:34 [1ms] [Warning] [omni.ext.plugin] [ext: .vscode] Extensions config 'extension.toml' doesn't exist '/home/nfdev/omniverse-playground/.vscode' or '/home/nfdev/omniverse-playground/.vscode/config'
2024-03-21 18:18:34 [1ms] [Warning] [omni.ext.plugin] [ext: scripts] Extensions config 'extension.toml' doesn't exist '/home/nfdev/omniverse-playground/scripts' or '/home/nfdev/omniverse-playground/scripts/config'
2024-03-21 18:18:34 [1ms] [Warning] [omni.ext.plugin] [ext: .git] Extensions config 'extension.toml' doesn't exist '/home/nfdev/omniverse-playground/.git' or '/home/nfdev/omniverse-playground/.git/config'
2024-03-21 18:18:34 [1ms] [Warning] [omni.ext.plugin] [ext: nucleus] Extensions config 'extension.toml' doesn't exist '/home/nfdev/omniverse-playground/nucleus' or '/home/nfdev/omniverse-playground/nucleus/config'
2024-03-21 18:18:34 [25ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.core-103.4.0+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-03-21 18:18:34 [25ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.usd-103.4.0+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-03-21 18:18:34 [27ms] [Warning] [omni.ext.plugin] [ext: omni.kit.widget.timeline-105.0.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-03-21 18:18:34 [28ms] [Warning] [omni.ext.plugin] [ext: omni.kit.window.sequencer-103.4.2-dev.3+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-03-21 18:18:34 [31ms] [Warning] [omni.ext.plugin] [ext: omni.paint.brush.attributes-1.3.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2024-03-21 18:18:34 [35ms] [Warning] [omni.ext.plugin] [ext: omni.usd.schema.sequence-2.3.0+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
[0.109s] [ext: omni.kit.async_engine-0.0.0] startup
[0.281s] [ext: omni.assets.plugins-0.0.0] startup
[0.281s] [ext: omni.stats-0.0.0] startup
[0.282s] [ext: omni.client-1.0.1] startup
[0.286s] [ext: omni.gpu_foundation-0.0.0] startup
[0.291s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.292s] [ext: carb.windowing.plugins-1.0.0] startup
2024-03-21 18:18:34 [289ms] [Warning] [carb.windowing-glfw.gamepad] Joystick with unknown remapping detected (will be ignored):  ASRock LED Controller [03000000ce260000a201000010010000]
[0.306s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 545.29.06     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA RTX A5000                 | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2231      | 568824bb.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.4 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.4, Kernel: 5.15.0-101-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i5-13600K | Cores: 14 | Logical: 28
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64059 | Free Memory: 59930
| Total Page/Swap (MB): 0 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|
[0.771s] [ext: omni.kit.pipapi-0.0.0] startup
[0.772s] [ext: omni.kit.pip_archive-0.0.0] startup
[... cut ...]
2024-03-21 18:18:37 [3,135ms] [Warning] [omni.kit.menu.utils.scripts.layout]   self._menu_placeholder = omni.kit.menu.utils.add_menu_items([MenuItemDescription(name="placeholder", show_fn=lambda: False)], name="Replicator", menu_index=90)
2024-03-21 18:18:37 [3,141ms] [Warning] [omni.kit.menu.utils.scripts.layout] Menu Replicator not found. Create with; ("menu_index" controls menu order)
2024-03-21 18:18:37 [3,142ms] [Warning] [omni.kit.menu.utils.scripts.layout]   self._menu_placeholder = omni.kit.menu.utils.add_menu_items([MenuItemDescription(name="placeholder", show_fn=lambda: False)], name="Replicator", menu_index=90)
[4.812s] Isaac Sim App is loaded.
```