isaac-sim/OmniIsaacGymEnvs #87: CartpoleCamera Task not running in debug mode

robinvetsch, 2023-10-20:
Hi all, 

I wanted to start the task "CartpoleCamera" from VSCode in debug mode. I get the following error:

[image]

The task without camera works as usual in debug mode. 

Many thanks for the help.

robinvetsch, 2023-10-23:
> Could you check again with the latest commit? This may be related to the other thread: #85.

I have used the latest commit, however the problem still exists.

kellyguo11, 2023-10-23:
It seems to be searching for the app file in the wrong directory from the error message `File doesn't exist: /home/robin/.local/share/ov/pkg/apps/omni.isaac.sim.python.gym.camera.kit`. We use a relative path to locate the app file, so perhaps the root directory changes when running in debug mode. You can try adding a cmdline argument `kit_app=/absolute/path/to/omni.isaac.sim.python.gym.camera.kit` to point to the absolute full path of the app file (https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/apps/omni.isaac.sim.python.gym.camera.kit)

Pipe-Runner, 2023-10-27:
A slightly unrelated question but why do we have `self._num_observations = 4` in CartpoleCamera example? Is that intended? 

@kellyguo11

kellyguo11, 2023-10-27:
Thanks for catching it, the `self._num_observations` shouldn't be 4, but the value was only used to initialize the observation buffer in the RLTask class, which was overridden by the CartpoleCamera implementation, so it didn't affect training. We will have this fixed.