isaac-sim/OmniIsaacGymEnvs #91: FrankaDeformable not working in 2023.1

Skylark0924, 2023-10-27:
## Bug
By running `[PYTHON_PATH] scripts/rlgames_train.py task=FrankaDeformable`, I got following error message:

```
...
Sim params does not have attribute:  physx
Sim params does not have attribute:  franka
Sim params does not have attribute:  beaker
Sim params does not have attribute:  cube
Pipeline:  GPU
Pipeline Device:  cuda:0
Sim Device:  GPU
Task Device: cuda:0
RL device:  cuda:0
2023-10-27 15:24:44 [21,326ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/envs/env_0/franka/panda_hand/visuals.proto_visuals_id0 that has not been populated
~*~*~*
~*~*~* Direct GPU Helper:
~*~*~*   1024 rigid dynamics
~*~*~*   1024 articulations, maxLinks=11
~*~*~*
2023-10-27 15:24:52 [29,322ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuSoftBodyView.cpp: 277
2023-10-27 15:24:52 [29,322ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuSoftBodyView.cpp: 274
2023-10-27 15:24:52 [29,322ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuSoftBodyView.cpp: 277
2023-10-27 15:24:52 [29,322ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuSoftBodyView.cpp: 274
2023-10-27 15:24:52 [29,322ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuSoftBodyView.cpp: 277
2023-10-27 15:24:52 [29,352ms] [Error] [omni.physx.tensors.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.tensors/plugins/gpu/GpuArticulationView.cpp: 71
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: Error executing job with overrides: ['task=FrankaDeformable', 'test=True', 'checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.0/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/franka_deformable.pth']

Error executing job with overrides: ['task=FrankaDeformable', 'test=True', 'checkpoint=omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.0/Isaac/Samples/OmniIsaacGymEnvs/Checkpoints/franka_deformable.pth']
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):

Traceback (most recent call last):
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/skylark/Github/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 150, in <module>

  File "/home/skylark/Github/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 150, in <module>
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:     
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: parse_hydra_configs()
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: 

    parse_hydra_configs()
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/main.py", line 94, in decorated_main

  File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/main.py", line 94, in decorated_main
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:     
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: _run_hydra(
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: 

    _run_hydra(
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra

  File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:     
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: _run_app(
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: 

    _run_app(
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457, in _run_app

  File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457, in _run_app
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]:     
2023-10-27 15:24:52 [29,374ms] [Error] [omni.kit.app._impl] [py stderr]: run_and_report(
2023-10-27 15:24:52 [29,375ms] [Error] [omni.kit.app._impl] [py stderr]: 

    run_and_report(
2023-10-27 15:24:52 [29,375ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 223, in run_and_report

  File "/home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
2023-10-27 15:24:52 [29,375ms] [Error] [omni.kit.app._impl] [py stderr]:     
2023-10-27 15:24:52 [29,375ms] [Error] [omni.kit.app._impl] [py stderr]: raise ex
2023-10-27 15:24:52 [29,375ms] [Error] [omni.kit.app._impl] [py stderr]: 
[... 122 lines cut ...]
2023-10-27 15:24:52 [29,377ms] [Error] [omni.kit.app._impl] [py stderr]: RuntimeError
2023-10-27 15:24:52 [29,377ms] [Error] [omni.kit.app._impl] [py stderr]: : 
2023-10-27 15:24:52 [29,378ms] [Error] [omni.kit.app._impl] [py stderr]: CUDA error: an illegal memory access was encountered
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

RuntimeError: CUDA error: an illegal memory access was encountered
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.
2023-10-27 15:24:52 [29,378ms] [Error] [omni.kit.app._impl] [py stderr]: 

2023-10-27 15:24:52 [29,381ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2023-10-27 15:24:52 [29,383ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2023-10-27 15:24:52 [29,384ms] [Error] [omni.physx.fabric.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 349
2023-10-27 15:24:52 [29,384ms] [Error] [omni.physx.fabric.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 352
2023-10-27 15:24:52 [29,384ms] [Error] [omni.physx.fabric.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 355
2023-10-27 15:24:52 [29,384ms] [Error] [omni.physx.fabric.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 358
2023-10-27 15:24:52 [29,384ms] [Error] [omni.physx.fabric.plugin] CUDA error: an illegal memory access was encountered: ../../../extensions/runtime/source/omni.physx.fabric/plugins/DirectGpuHelper.cpp: 361
2023-10-27 15:24:52 [29,388ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-10-27 15:24:52 [29,433ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-10-27 15:24:52 [29,442ms] [Warning] [omni.core.ITypeFactory] Module /home/skylark/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```
---
## Trails
1. Other tasks (like FrankaCabinet) work well in 2023.1. So I think it should be an issue about the deformable simulation.
2. Since the error shows `[Error] [omni.physx.fabric.plugin]`, I have tried to launch isaac sim to open and close these extensions, but nothing changed.
[image]

kellyguo11, 2023-10-27:
Hi there, can you try lowering the number of environments with the `num_envs` flag? It's possible the machine ran out of memory during the deformable simulation.

Skylark0924, 2023-10-28:
Thanks for your reply. I have tried to modify the num_envs to 64/32/16, the error keeps the same

dhruvkm2402, 2023-10-30:
Hi @Skylark0924 , It worked for me with the default num_envs settings. 
I have a workstation though with RTX A6000 and 256GB RAM

XEL-Maker, 2023-11-14:
Hi @kellyguo11

> Hi there, can you try lowering the number of environments with the `num_envs` flag? It's possible the machine ran out of memory during the deformable simulation.

I tried above solution but it's not worked well.(Laptop with RTX 4070 and 32GB RAM, num_envs: 4 and 8 tried)
The env is well setted and after it crash.
It also happens at extension workflow.(clicking start after loading env)
I treid it wih default setting at my workstation though with RTX A4000 and 32GB RAM it worked.
Both are Window 11 OS.
Including log.
[kit_20231114_092711.log](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/files/13343220/kit_20231114_092711.log)