isaac-sim/OmniIsaacGymEnvs #94: cartpole camera headless

domhuh, 2023-10-30:
CartpoleCamera seems to work with headless=False, verified by the exportImages=True.
However, with headless=True, the image tensor is None.
Is there a fix to this?

kellyguo11, 2023-11-03:
Hi there, the CartpoleCamera example should be able to run with `headless=True` as well. Are you seeing any errors being reported during training?

domhuh, 2023-11-03:
Only warnings are being reported:
2023-11-03 17:18:19 [251,488ms] [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
2023-11-03 17:18:25 [257,216ms] [Warning] [omni.fabric.plugin] No source has valid data array=00000216525FE938 usdValid=0 cpuValid=0
2023-11-03 17:18:25 [257,222ms] [Warning] [omni.fabric.plugin] No source has valid data array=00000217B92CD550 usdValid=0 cpuValid=0
2023-11-03 17:18:25 [257,856ms] [Warning] [carb] Client omni.hydratexture.plugin has acquired [carb::settings::ISettings v1.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2023-11-03 17:18:26 [258,001ms] [Warning] [carb] Client omni.hydratexture.plugin has acquired [carb::dictionary::IDictionary v1.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2023-11-03 17:18:34 [266,668ms] [Warning] [carb] Client omni.kit.viewport.legacy_gizmos has acquired [carb::settings::ISettings v1.0] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
2023-11-03 17:18:34 [266,670ms] [Warning] [carb] Client omni.kit.viewport.legacy_gizmos has acquired [carb::dictionary::IDictionary v1.1] 100 times. Consider accessing this interface with carb::getCachedInterface() (Performance warning)
~*~*~*
~*~*~* Direct GPU Helper:
~*~*~*   32 articulations, maxLinks=3
~*~*~*
Image tensor is NONE!

The only change I made was to the headless: True and exportImages: True.

domhuh, 2023-11-03:
Sorry I didn't mean to close.