isaac-sim/OmniIsaacGymEnvs #96: Bottom drawer in Franka Cabinet cannot be opened

Pipe-Runner, 2023-11-04:
Hi, I am making an obvious mistake. Still, I have changed the observation buffer and modified the Cabinet view to be able to interact with and open the bottom drawer using Franka. The robot seems to reach and even try to grip and pull the drawer, but the drawer doesn't move even a little. I might start digging into the USD, but it would be nice if you could give me a heads-up if there is something I need to change in the USD itself for it to work.

Also, while in "play" mode in Isaac's sim in Omniverse, I try to apply force on entities like the cabinet as shown in different videos on YouTube, but I can't seem to do it using the mouse. Is there a key combination I should press to interact with objects directly instead of selecting them?

sujitvasanth, 2023-11-04:
in isaac Sim in play mode select the prim you would like to move, hold down the shift key and drag with the left mouse button. You will see the green line representing the force vector. If you see the green lines but the object is not moving - zoom out and make the green line longer as it seems the force is proportionate to the line length.

[image]

Pipe-Runner, 2023-11-04:
> in isaac Sim in play mode select the prim you would like to move, hold down the shift key and drag with the left mouse button. You will see the green line representing the force vector. If you see the green lines but the object is not moving - zoom out and make the green line longer as it seems the force is proportionate to the line length.
> 
> [image]
> 
> Regarding draw opening - I think in earlier versions I remember seeing in the video demos there was some cheats in the coding to overcome the physics of handilng the draw opening...suppose this is probably still the case

Thankyou for taking out the time to reply to my questions. I'll try this out to check if the bottom drawer can indeed be interacted and opened.

Pipe-Runner, 2023-11-04:
Update: It seems like in the sandbox, I am opening the drawer by applying forces using the mouse. I am not sure what's going on in the RL training.

sujitvasanth, 2023-11-06:
Hi @Pipe-Runner you might find this video of some help
https://www.nvidia.com/en-us/on-demand/session/gtcfall21-a31118/
look at 39mins 30s in as they demonstrate the Franka cabinet task - the draw has zero friction so that it falls open even with a gentle pull - it's not an example I had a particular interest in so I didn't really want to delve any further...