isaac-sim/OmniIsaacGymEnvs #97: anymal_instanceable.usd

SengHongLEE, 2023-11-04:
How can I get "../Isaac/Robots/ANYbotics/anymal_instanceable.usd" without accessing the localhost server?

sujitvasanth, 2023-11-04:
you can navigate in the IsaacSim content browser to
omniverse://localhost/NVIDIA/Assets/Isaac/2023.1.0/Isaac/Robots/
or the 2022.2.1 equivilent

[image]

right click and download.. I have included a zip of the files you wanted
[ANYbotics.zip](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/files/13257503/ANYbotics.zip)

SengHongLEE, 2023-11-05:
Thank lots for your responses on time. I have reached the file through localhost with your clear tutorial.

closed 2023-11-05 by SengHongLEE