isaac-sim/OmniIsaacGymEnvs #98: Sometimes, I encounter segfault. Not sure what's happening.

nikepupu, 2023-11-05:
Fatal Python error: Segmentation fault

Current thread 0x00007fa8c84e1b80 (most recent call first):
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/extsPhysics/omni.physx.ui-105.1.9-5.1/omni/physxui/scripts/physxProgressView.py", line 47 in _on_progress_settings_changed
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 479 in render
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 561 in play
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 387 in initialize_physics
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 408 in reset
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/world/world.py", line 282 in reset
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.gym/omni/isaac/gym/vec_env/vec_env_base.py", line 126 in set_task
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/envs/vec_env_rlgames.py", line 47 in set_task
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/utils/task_util.py", line 105 in initialize_task
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 121 in parse_hydra_configs
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/core/utils.py", line 186 in run_job
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119 in run
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458 in <lambda>
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220 in run_and_report
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457 in _run_app
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394 in _run_hydra
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/main.py", line 94 in decorated_main
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 150 in <module>

kellyguo11, 2023-11-06:
Hi there, could you provide more information on which environment you are running and the command you are using?

sujitvasanth, 2023-11-06:
hi @nikepup your error message is truncated so its not possible for us to advise. The segmentation fault usually implies something has gone wrong with your handling of the USD or the usd file itself. I usually find tracing the end of the message nest in your code usually gives a good clue to what is happening - please could you provide more context?

nikepupu, 2023-11-06:
hi @sujitvasanth this is actually the full error message. 

fps step: 3574 fps step and policy inference: 3501 fps total: 3320 epoch: 1248/15000 frames: 5107712
fps step: 3485 fps step and policy inference: 3419 fps total: 3276 epoch: 1249/15000 frames: 5111808
fps step: 3471 fps step and policy inference: 3406 fps total: 3267 epoch: 1250/15000 frames: 5115904
fps step: 3432 fps step and policy inference: 3368 fps total: 3233 epoch: 1251/15000 frames: 5120000
Fatal Python error: Segmentation fault

Thread 0x00007fd7656a1700 (most recent call first):
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 320 in wait
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/multiprocessing/queues.py", line 231 in _feed
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 953 in run
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fd77eee8700 (most recent call first):
  <no Python frame>

Thread 0x00007fd76cffd700 (most recent call first):
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/selectors.py", line 416 in select
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/multiprocessing/connection.py", line 931 in wait
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/multiprocessing/connection.py", line 424 in _poll
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/multiprocessing/connection.py", line 257 in poll
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/multiprocessing/queues.py", line 113 in get
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/tensorboardX/event_file_writer.py", line 202 in run
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fe831d4eb80 (most recent call first):
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/physics_context/physics_context.py", line 571 in _step
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/simulation_context/simulation_context.py", line 466 in step
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core/omni/isaac/core/world/world.py", line 380 in step
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/envs/vec_env_rlgames.py", line 64 in step
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/utils/rlgames/rlgames_utils.py", line 104 in step
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 519 in env_step
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 752 in play_steps
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 1182 in train_epoch
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 1318 in train
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 116 in run_train
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 133 in run
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 74 in run
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 142 in parse_hydra_configs
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/core/utils.py", line 186 in run_job
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119 in run
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458 in <lambda>
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220 in run_and_report
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457 in _run_app
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394 in _run_hydra
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/hydra/main.py", line 94 in decorated_main
  File "/home/nikepupu/Desktop/OmniIsaacGymEnvs/omniisaacgymenvs/scripts/rlgames_train.py", line 150 in <module>

[... cut ...] ive, scipy.optimize._bglu_dense, scipy.optimize._lsap, scipy.optimize._direct, scipy.interpolate._bspl, scipy.interpolate._ppoly, scipy.interpolate.interpnd, scipy.interpolate._rbfinterp_pythran, scipy.interpolate._rgi_cython (total: 99)
/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/python.sh: line 41: 67578 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python

I observed it happens around the time tunneling happens during the physics simulation.

nikepupu, 2023-11-07:
Hi it's a custom opening drawer task with a mobile robot and kinova arm and robotiq 2f85 finger. The usds are generated using assets from partnet mobility dataset. the tuneling happens when try to open the drawer. I am using a modifed objective during RL that the robot only get reward if opening the drawer using handle. The collision meshes are quite simple after applying https://github.com/SarahWeiii/CoACD. I did some digging,  it seems physx will only generate contact once per triangle. What i observed is that the arm will go through the front panel and go into the drawer without opening it. After that happens for a while, the RL training will crash.  I will run the code more time tomorrow using modified meshes after applying subdivision.

nikepupu, 2023-11-08:
@kellyguo11 are substeps still useful in yaml config files?

nikepupu, 2023-11-21:
a bit more update, it seems this is an issue exclusive to non headless mode.

nikepupu, 2023-11-21:
Ok problem solved. This is related to file descriptor limits. Increasing the max number of files allowed to open following : https://docs.omniverse.nvidia.com/dev-guide/latest/linux-troubleshooting.html#to-increase-the-file-descriptor-limit  solves the issue.

nikepupu, 2023-11-22:
The problem still exist.

But I can confirm this only happens in GUI mode. The problem disappears when running the headless mode.

nikepupu, 2023-12-18:
Ok, now I can confirm this issue is related to GPU dynamics. switching to CPU collisions solves the problem, however the simulation is a lot slower.

nikepupu, 2024-01-18:
Hi, my workaround is to use CPU simulation.  There is definitely a bug in GPU simulation.