isaac-sim/OmniIsaacGymEnvs #100: Need comments to understand magic numbers

Pipe-Runner, 2023-11-07:
I understand this is not an issue, but using cryptic code and magic numbers without documentation or comments makes it very hard to modify the tasks. I maybe over stepping there since perhaps I am the one who is unable to figure the code out. So feel free to point me in the right direction if that happens to be the case.

In Franka Cabinet, in `init_data`, what exactly is this magic number array and how am I supposed to find it for a different scenario? 
https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/8cf773ab6cac0c8e0d55f46d6d7d258e781c6458/omniisaacgymenvs/tasks/franka_cabinet.py#L228

Also, is there any documentation that explains the implementation?

Pipe-Runner, 2023-11-09:
Okay, I may have just figured it out.
[image]
Those magic numbers are the local transform of the grasp position you would want the robot to have. Since this is a relative transform, every time observation is computed, you combine it with the transformation of the drawer itself so this pose keeps moving with the drawer, and the robot tries to reduce the distance between its grasp and this desired grasp position. If I can translate this video, I will probably help document the code with the maths.

This video explains it further, but it is in Korean.

kellyguo11, 2023-11-09:
Hi there, you are correct on the explanation of the `drawer_local_grasp_pose` values. We will try to provide better documentation/commenting for the examples.