isaac-sim/OmniIsaacGymEnvs #101: Joint Velocities Incorrect?

davesarmoury, 2023-11-08:
I am running a simulation task using OmniIsaacGymEnvs. The simulation has reached steady-state and the robot is stationary (no joints are visibly moving).

Run the Anymal task from this branch in test mode: [j_vel](https://github.com/davesarmoury/OmniIsaacGymEnvs/tree/j_vel)

Calling get_joint_velocities() on the ArticulationView returns non-zero values (+/- 0.3). Why would this be? Are the joint velocities relative to the joint itself, or relative to some other frame? Are they in radians per second, or degrees for some reason?  The current targets are being set using set_joint_position_targets()

I'm using 2023.2.1

kellyguo11, 2023-11-09:
Hi there, thanks for the report. We will look into this.

AndrePatri, 2023-11-16:
Hi @davesarmoury, @kellyguo11 I am having the same issue on IsaacSim2023.1.0 (and I remember having it also on the 2022 version). It might be some IsaacSim bug though, since I am not using OmniIsaacGymEnvs at all and getting nonzero velocity when the robot I am using is stationary (if I get the joint). This happens for all kinds of velocity (e.g. linear and joints).
For example:
[image]
and
[image]
These are graphs of the quantities I am getting from IsaacSim, linear velocity of my root link and joint velocities of the links.
As you see, even if the positions are clearly stationary, velocities are non-null

AndrePatri, 2023-11-16:
Also, going through IsaacSim discussions I spotted [this](https://forums.developer.nvidia.com/t/dof-velocity-offset-at-rest/205799/15), where they say this is an expected behaviour. It is not clear to me however how this data can then be used for control purposes.

kellyguo11, 2023-11-20:
Hi there, there are a few parameters that you can try tuning to achieve more accurate velocities. Decreasing the simulation dt can help with numerical issues when the velocities converge, which is likely the main issue in this case. Other parameters include increasing the position iteration count, or trying with a significantly larger stiffness value to avoid oscillations.

AndrePatri, 2023-11-21:
Hi @kellyguo11, thanks for the feedback. 
I already tried to decrease the integration_dt (I was using 0.01 and I lowered it to 0.001). However, from what I saw, this didn't make any difference. 
From the documentation I can see methods for setting the iteration count on Articulations/ArticulationViews on both positions and velocities. Should I try to play also with the velocity one? What is their meaning?

kellyguo11, 2023-11-21:
You can try setting velocity iteration to 0, since the velocity iterations will tamper with the velocities, but those updates won't be reflected in the positions. Our physics team is also working on a set of recommendations for different applications for working with the velocities, which hopefully we'll be able to share soon.

AndrePatri, 2023-11-21:
Thanks @kellyguo11. 
I think for robotics applications it's crucial for all the quantities you get from the simulation.
Having for example this issue with the velocity without realizing it exists can completely alter the personal of any controller, including RL ones. At least a warning on this in the documentation of the getter methods is imperative, in my opinion.

bakshienator77, 2025-09-13:
I am facing this issue in isaacsim 5.0.0 too. 

[image]