isaac-sim/OmniIsaacGymEnvs #102: Question about Docker setting

doyeon-danny, 2023-11-08:
Hi, I am setting Docker container to launch Omni Isaac Gym project only in Docker. 

1. Docker container
 I made a container with `run_docker_viewer.sh` file with erasing `--rm` to remain this container.
[image]

2. Custom RL example
I made `cartpole_task.py` and `cartpole_train.py` following below tutorial.
[https://docs.omniverse.nvidia.com/isaacsim/latest/isaac_gym_tutorials/tutorial_gym_isaac_gym_new_oige_example.html](url)
And I got this error messages.
[image]

3. IsaacGym Docker (it is different with 1st docker)
I guess `2023.1.0-hotfix.1` image(that is used to make a Docker container) doesn't have Cuda, nvcc etc.
When I downloaded played IsaacGym Docker, it works with nvcc because its' dockerfile uses `FROM nvcr.io/nvidia/pytorch:21.09-py3` base image.

4. Then how can I develop OmniIsaacGym fully in Docker?
I found Dockerfile Multi Stage that can make a docker image with multiple docker images.
Is it the correct way to construct Full Docker environment?

I hope you can read and answer my questions, thank you.

kellyguo11, 2023-11-09:
Hi there, the Isaac Sim docker containers will use the cuda runtime that's included in the drivers of the host machine. Please make sure you have a valid driver installed and that the machine is capable of running with rendering if using `run_docker_viewer.sh`. For a headless machine, you can use the headless docker script `run_docker.sh`. It looks like your task may be running into an error, `AttributeError: 'CartpoleTask' object has no attribute 'reset'`, which could also be the cause of the CUDA errors. To validate this, you can try running one of the built-in environments using the docker container.

doyeon-danny, 2023-11-10:
Hi, I appreciate your answer.
IsaacGym has 11.4 Cuda version, then is it better to install 11.4 Cuda?
Doesn't it matter what cuda version is?

kellyguo11, 2023-11-10:
Hi there, yes that should work. As long as you have one of the recommended driver versions installed, isaac sim should be able to use the cuda runtime that comes with the driver. I don't think a separate cuda toolkit installation is required.