isaac-sim/OmniIsaacGymEnvs #104: Launching Isaac Sim in extention workflow takes painfully long and failed

Guanbin-Huang, 2023-11-10:
i run this:
./isaac-sim.gym.sh --ext-folder /home/lr/OmniIsaacGymEnvs/omniisaacgymenvs

[image]

[image]

[image]

Guanbin-Huang, 2023-11-10:
i also tried this.

[image]

Guanbin-Huang, 2023-11-10:
i also tryied this 

./isaac-sim.gym.sh --ext-folder /home/lr/OmniIsaacGymEnvs

[image]

[image]

[image]

[image]

kellyguo11, 2023-11-10:
Hi there, the extension name should be `OmniIsaacGymEnvs` (camel case), which should be fixed in the latest isaac sim 2023.1.0-hotfix.1 release. The `--ext-folder` flag should point to the parent directory of `OmniIsaacGymEnvs`, which in your case would be `/home/lr`

Guanbin-Huang, 2023-11-11:
I follow your advice and I run the following command:
[image]

after i modify the omni.isaac.sim.gym.kit, i use the pip install -e . (not sure it's used for updating the pkg?)
[image]

then the log said
[image]
[image]
[image]
[image]

kellyguo11, 2023-11-12:
Hi there, please use the Isaac Sim 2023.1.0-hotfix.1 release with the latest OmniIsaacGymEnvs repo (2023.1.0b), or with the Isaac Sim 2023.1.0 release (not including the hot fix), you can use the OmniIsaacGymEnvs 2023.1.0a release. You can try adding back the `"psutil==5.9.3"` dependency to setup.py, which should not be needed anymore with the hotfix Isaac Sim release.

Guanbin-Huang, 2023-11-12:
[image]
[image]
I already uninstall the old version and install the latest version(this is what you name "Hot Fix" isaac sim release.) and I git clone the lastest version of omniverse isaac gym by 
git clone https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs.git

Guanbin-Huang, 2023-11-12:
even i'm using the the latest version. I still got the errors:

[image]

No module named "psutils and six"

[image]

Guanbin-Huang, 2023-11-12:
I managed to fix the bug in a very hardcode way as followed
for each failed script due to lack of psutil or six

1. first i make sure which python env i'm using for isaac sim-2023.1.0-hotfix.
and i add the path of python packages into all failed script like the following

[image]

[image]
let's say the file saying no module xxxx is extests.py. 

i desire a more scientific solution to fix it.

XEL-Maker, 2023-11-13:
@kellyguo11
Hi 
I face with the same problem as like above.
I'm using Window 11 with Isaac Sim 2023.1.0-hotfix.1.
I've tried bot 2023.1.0 and 2023.1.0-hotfix.1 and the result was 2023.1.0 can not open 
2023.1.0-hotfix.1 is same as above image wiht "no examlple button"
Below is the error at 2023.1.0.
[image]

Guanbin-Huang, 2023-11-13:
> @kellyguo11 Hi I face with the same problem as like above. I'm using Window 11 with Isaac Sim 2023.1.0-hotfix.1. I've tried both 2023.1.0 and 2023.1.0-hotfix.1 and the result was 2023.1.0 can not open 2023.1.0-hotfix.1 is same as above image wiht "no examlple button" Below is the error at 2023.1.0. [image]

follow what i've done. you can fix it in a not so decent way.

kellyguo11, 2023-11-13:
Does it work for you if you add the missing dependencies, such as `"psutil==5.9.3"` into setup.py and re-run `pip install -e .`? or try running `pip install -e . --force-reinstall` to make sure all of the required libraries get installed.

XEL-Maker, 2023-11-14:
@kellyguo11
I solved it using the solution you suggested.
For some reason, it seems that the Python specified by doskey was not 3.10.
I think the cause of the problem is that multiple versions of Isaac Sim and Python are installed.
My problem has been solved.
@Guanbin-Huang I think my problem was a little different from yours.
After correcting the Python path and reinstalling, the OIGE package was additionally installed and began to operate normally.
My issue has been resolved.
Please close if your issue has been resolved.

Guanbin-Huang, 2023-11-14:
> @kellyguo11 I solved it using the solution you suggested. For some reason, it seems that the Python specified by doskey was not 3.10. I think the cause of the problem is that multiple versions of Isaac Sim and Python are installed. My problem has been solved. @Guanbin-Huang I think my problem was a little different from yours. After correcting the Python path and reinstalling requirements, the OIGE package was additionally installed and began to operate normally. My issue has been resolved. Please close if your issue has been resolved.

actually, my issue is not yet solved. Cuz now i used hard coding. Not a general solution.