isaac-sim/OmniIsaacGymEnvs #106: anymal.usd issue

SengHongLEE, 2023-11-15:
I would like to add a foot contact sensor on the foot (mesh_1) respectively, unfortunately it can't be done due to the exist of the 'visuals', how can I deal with it alternatively? Thanks in advanced.
[image]

dhajnes, 2023-11-16:
Hi, I believe you are in the "instancable" asset USD. If you want to add contact sensor, you can go into the anymal_base.usd and do it there, although instancable asset will be the one you want to load. My suggestions are:
a) just put the contact sensor on the LF_SHANK prim, not actually on the mesh.
b) look into the example tasks of `anymal.py` and `anymal_terrain.py`. One of them is actually preparing the model for contacts programmatically, so you might not need to edit the USD at all. 

Hope this helps.

dhajnes, 2023-11-17:
> where is the exact location of the model for contacts? I can't find it.

`prepare_contacts()` at the bottom here:

https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/robots/articulations/anymal.py

Andrej

SengHongLEE, 2023-11-22:
Btw, how could I get the torque value of each joint of _anymals （an object of AnymalView）?

dhajnes, 2023-11-23:
Get inspired from the task `anymal_terrain`. They use the position control
mode, so they compute the torque based on the default joint positions, the
stiffness and damping of the joint drives etc. Also see the
`articulation_view.py` in 'omni.isaac.core` library. There you will find
functions such as `get_joint_forces()` and `get_joint_efforts()` I believe.

SengHongLEE, 2023-11-27:
> `prepare_contacts()` at the bottom here: https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/main/omniisaacgymenvs/robots/articulations/anymal.py Andrej
> […](#)

Regarding this, I can't fully understand the content of this function, where may I get through this (i.e. does there have some detailed tutorials or documentations to get the feedback of the robot's foot contact situation with the ground) ?

dhajnes, 2023-11-28:
Hi, its a bit more complicated. The answer to your questions is `anymal_view.py`. You need to create a RigidPrimView() where the track_contact_forces = True and prepare_contact_sensors=True. You don't set it there directly, make sure the boolean variables `track_contact_forces` and `prepare_contact_sensors` are set to True by putting `anymal.prepare_contacts(self._stage, anymal.prim)` in `get_anymal()` function in `omniisaacgymenvs/tasks/anymal.py` and `track_contact_forces=True` in the `set_up_scene()` in `tasks/anymal.py`, as written in the second snippet below.

This is a snippet from `omniisaacgymenvs/robots/articulations/views/anymal_view.py`:

```python
self._knees = RigidPrimView(
            prim_paths_expr="/World/envs/.*/anymal/.*_THIGH",
            name="knees_view",
            reset_xform_properties=False,
            track_contact_forces=track_contact_forces,
            prepare_contact_sensors=prepare_contact_sensors,
        )
        self._base = RigidPrimView(
            prim_paths_expr="/World/envs/.*/anymal/base",
            name="base_view",
            reset_xform_properties=False,
            track_contact_forces=track_contact_forces,
            prepare_contact_sensors=prepare_contact_sensors,
        )
```
        
You need to add a RigidPrimView() to the prim (primitive) you want to attach the force sensor to add the view to the scene as is done in `tasks/anymal.py` in `setup_scene()`

as:

```python
def set_up_scene(self, scene) -> None:
        self.get_anymal()
        super().set_up_scene(scene)
        self._anymals = AnymalView(prim_paths_expr="/World/envs/.*/anymal", name="anymalview", track_contact_forces=True)
        scene.add(self._anymals)                                # <<< like these
        scene.add(self._anymals._knees)                   # <<< like these
        scene.add(self._anymals._base)

        return
```

Then you can access the forces of the given rigid prim view as is written for example in `tasks/anymal_terrain.py` in function `check_termination()`:

```python
   ...
   knee_contact = (
            torch.norm(self._anymals._knees.get_net_contact_forces(clone=False).view(self._num_envs, 4, 3), dim=-1)
            > 1.0
        )
    ...
```

The documentation is not good enough, tutorials are covering only very basic usage and I just sort of searched all the repos until I found what I was looking for... So checking the `omni.isaac.core` folder where you have the omni libraries installed will help you with some function definitions on `rigid_prim_view` and `rigic_contact_view` etc.

Hope this helps.

SengHongLEE, 2023-12-13:
Sorry for late response.  I just want to put a contact sensor on each foot of Unitree-A1 instead of whole leg, so I have modified the prepare_contacts() function like this. 

`    def prepare_contacts(self, stage, prim):
        for link_prim in prim.GetChildren():
            if link_prim.HasAPI(PhysxSchema.PhysxRigidBodyAPI):
                if "_foot" in str(link_prim.GetPrimPath()):    #the change
                    rb = PhysxSchema.PhysxRigidBodyAPI.Get(stage, link_prim.GetPrimPath())
                    rb.CreateSleepThresholdAttr().Set(0)
                    cr_api = PhysxSchema.PhysxContactReportAPI.Apply(link_prim)
                    cr_api.CreateThresholdAttr().Set(0) `
Besides, I have built a RigidPrimView like this:

`        self._foots = RigidPrimView(
            prim_paths_expr="/World/envs/.*/a1_description/.*_foot",
            name="foots_view",
            reset_xform_properties=False,
            track_contact_forces=track_contact_forces,
            prepare_contact_sensors=prepare_contact_sensors,
        )`

I have set the relevant parameters track_contact_forces as True but prepare_contact_sensors remain as False. I got this error:
[image]

If I change prepare_contact_sensors as True, it works. Yet, I don't think the return of 'self._anymals._foots.get_net_contact_forces(clone=False)' is what i want because the shape of tensor is [num_envs*4, 3], seems that it still returns the sensor feedback of whole leg.

Actually, my final goal is to get the feedback of the foot contact with the ground in boolean type (i.e. if foot contact with ground return 1, vice versa.)

SengHongLEE, 2023-12-13:
Sorry for late response.  I just want to put a contact sensor on each foot of Unitree-A1 instead of whole leg, so I have modified the prepare_contacts() function like this. 

[image]

Besides, I have built a RigidPrimView like this:

[image]

I have set the relevant parameters track_contact_forces as True but prepare_contact_sensors remain as False. I got this error:
[image]

If I change prepare_contact_sensors as True, it works. Yet, I don't think the return of 'self._anymals._foots.get_net_contact_forces(clone=False)' is what i want because the shape of tensor is [num_envs*4, 3], seems that it still returns the sensor feedback of whole leg.

Actually, my final goal is to get the feedback of the foot contact with the ground in boolean type (i.e. if foot contact with ground return 1, vice versa.)