isaac-sim/OmniIsaacGymEnvs #107: How do I retrieve vectorized contact forces from RigidPrimView's get_net_contact_forces()?

AndrePatri, 2023-11-16:
Hi, 
I am trying to retrieve contact forces from IsaacSim2023.1.0's simulation of my robot.
I can get this info by attaching a contact sensor to each contact link. However, this approach scales terribly when the number of environments grow. 
As a consequence, I was looking for a vectorized way of getting contact info and thanks to [https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/31](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/31), I found out that it's possible to use RigidPrimView to encapsulate contact prims across environments.
The issue I am facing is that it reports the net impulse on the contact links, scaled by dt and, as  a consequence, when the robot is still I am not able to get the contact force. 
Is there a way of overcoming this? Am I missing something?

Milad-Rakhsha, 2023-11-20:
Can you elaborate a bit on why you aren't able to get contact forces? Do you need net contact forces per link or do you need to access individual contact data between different links?

AndrePatri, 2023-11-21:
Hi @Milad-Rakhsha. So basically, I am running an MPC on a wheeled quadruped robot. I would need the contact forces (ideally, normal + modulus or directly the components) on some specific links, in particular the feet.
When I use the net_contact_forces method, I can only get non-null values when the robot is spawned, but that's because it falls towards the ground for a couple of centimeters and this produces a vertical impulse. Then, as soon as the robot stabilizes, I am always reading zero from the method. 
However, as soon as I activate my MPC (with the robot still), I am indeed able to read a non null vertical impulse at each contact foot (with total value amouting to the robot's weight), but I think that's because the MPC is introducing some noise into the commanded torques and this is triggering contact impulses with the ground.
As I side note, with the MPC active, I can retrieve reasonable values of the vertical contact force, but the ones along the plane (x, y components) are basically always zero, even if the robot is rolling aggressively on the ground.
Hopefully this gives you a bit more context on the issue.