isaac-sim/OmniIsaacGymEnvs #108: headless

1725917163, 2023-11-16:
Hello, we found a bug during model inference. When the headless parameter settings are different, the results of model inference are the same. The main reason is that the first frame state variable returned by the Isaac-sim simulator is inconsistent. In theory, it should are the same, is this normal?

kellyguo11, 2023-11-20:
Hi there, we cannot guarantee determinism between headless and non-headless modes. It is expected that the initial states in non-headless mode can be slightly different than the states in headless mode, but the overall simulation behaviour should be similar.

guin18, 2024-01-11:
I have the same problem as @1725917163 's, so I tried to train the model with headless=False. The performance of the model is not so good as the one with headless=True, and is far from convergence.