isaac-sim/OmniIsaacGymEnvs #112: Are there controls implemented based on the pytorch?

xialuanshi, 2023-11-21:
Hello, when I use OIGE, I want to build a task-level RL model. I will use some controller files in Isaac-sim, but some controllers only support the calculation of np data. How should I perform related conversions? Or are there controllers implemented based on the pytorch?

kellyguo11, 2023-11-21:
Hi there, for numpy-only controllers, you can convert the torch tensors into numpy arrays with `tensor.cpu().numpy()`.

xialuanshi, 2023-11-22:
> Hi there, for numpy-only controllers, you can convert the torch tensors into numpy arrays with `tensor.cpu().numpy()`.

Yes, this is a method, but I want to put all the calculations on the GPU. Otherwise, during the process of docking with isaac-sim, I need to convert the data into numpy first, and then complete the calculation. After obtaining the results, It needs to be converted into a pytorch tensor, and then let Isaac-sim simulate it.