isaac-sim/OmniIsaacGymEnvs #115: How to import assets from URDF files without user interface?

damianliumin, 2023-11-26:
Hi! I am running code on a headless server. I find that existing methods to import urdf files listed in 
[https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_import_urdf.html](https://github.com/NVIDIA-Omniverse/urdf-importer-extension) 
and 
[https://github.com/NVIDIA-Omniverse/urdf-importer-extension](https://github.com/NVIDIA-Omniverse/urdf-importer-extension)
cannot be used without user interface. Is there any way to import assets from URDF files in this situation?

kellyguo11, 2023-11-28:
HI there, please checkout out https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_import_urdf.html#importing-urdf-using-python for the python interface for the URDF importer.

damianliumin, 2023-11-30:
Thanks! When I tried this code with the provided python executable, it cannot recognize `from omni.importer.urdf import _urdf`.

closed 2023-12-23 by damianliumin

Casselll, 2024-11-28:
> HI there, please checkout out https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_advanced_import_urdf.html#importing-urdf-using-python for the python interface for the URDF importer.

Hi there, I tried this but turned out it still need to start ui to make it work, any other methods will be great thx~