isaac-sim/OmniIsaacGymEnvs #120: Segmentation fault in Docker

Cdfghglz, 2023-12-15:
Hi, I am unable to run the isaac-sim inside Docker

```bash
cdf@xps15:~/ws/pr_omni_isaac_gym_envs$  (main)./docker/run_docker.sh 
docker: Error response from daemon: Conflict. The container name "/isaac-sim-oige" is already in use by container "46783acb7a7b1cfbe70fc53989462c3ac653a0be1aea47989ed16e1da589b841". You have to remove (or rename) that container to be able to reuse that name.
See 'docker run --help'.
Obtaining file:///workspace/omniisaacgymenvs
Requirement already satisfied: numpy==1.23.5 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from omniisaacgymenvs==2023.1.0b0) (1.23.5)
Requirement already satisfied: protobuf==3.20.2 in /isaac-sim/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (3.20.2)
Requirement already satisfied: omegaconf==2.3.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (2.3.0)
Requirement already satisfied: hydra-core==1.3.2 in /isaac-sim/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (1.3.2)
Requirement already satisfied: urllib3==1.26.16 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from omniisaacgymenvs==2023.1.0b0) (1.26.16)
Requirement already satisfied: rl-games==1.6.1 in /isaac-sim/kit/python/lib/python3.10/site-packages (from omniisaacgymenvs==2023.1.0b0) (1.6.1)
Requirement already satisfied: packaging in /isaac-sim/exts/omni.isaac.core_archive/pip_prebundle (from hydra-core==1.3.2->omniisaacgymenvs==2023.1.0b0) (23.0)
Requirement already satisfied: antlr4-python3-runtime==4.9.* in /isaac-sim/kit/python/lib/python3.10/site-packages (from hydra-core==1.3.2->omniisaacgymenvs==2023.1.0b0) (4.9.3)
Requirement already satisfied: PyYAML>=5.1.0 in /isaac-sim/exts/omni.pip.compute/pip_prebundle (from omegaconf==2.3.0->omniisaacgymenvs==2023.1.0b0) (6.0.1)
Requirement already satisfied: tensorboard<3.0.0,>=2.8.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.15.1)
Requirement already satisfied: opencv-python<5.0.0,>=4.5.5 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (4.8.1.78)
Requirement already satisfied: tensorboardX<3.0,>=2.5 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.6.2.2)
Requirement already satisfied: wandb<0.13.0,>=0.12.11 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.12.21)
Requirement already satisfied: psutil<6.0.0,>=5.9.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (5.9.6)
Requirement already satisfied: setproctitle<2.0.0,>=1.2.2 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.3.3)
Requirement already satisfied: gym[classic-control]<0.24.0,>=0.23.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.23.1)
WARNING: gym 0.23.1 does not provide the extra 'classic-control'
Requirement already satisfied: cloudpickle>=1.2.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from gym[classic-control]<0.24.0,>=0.23.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.0.0)
Requirement already satisfied: gym-notices>=0.0.4 in /isaac-sim/kit/python/lib/python3.10/site-packages (from gym[classic-control]<0.24.0,>=0.23.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.0.8)
Requirement already satisfied: tensorboard-data-server<0.8.0,>=0.7.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.7.2)
Requirement already satisfied: grpcio>=1.48.2 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.60.0)
Requirement already satisfied: google-auth<3,>=1.6.3 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.25.2)
Requirement already satisfied: requests<3,>=2.21.0 in /isaac-sim/exts/omni.pip.cloud/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.31.0)
Requirement already satisfied: werkzeug>=1.0.1 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.0.1)
Requirement already satisfied: google-auth-oauthlib<2,>=0.5 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.2.0)
Requirement already satisfied: setuptools>=41.0.0 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (68.0.0)
Requirement already satisfied: markdown>=2.6.8 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.5.1)
Requirement already satisfied: absl-py>=0.4 in /isaac-sim/kit/python/lib/python3.10/site-packages (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.0.0)
Requirement already satisfied: six>1.9 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.16.0)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /isaac-sim/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.3.0)
Requirement already satisfied: cachetools<6.0,>=2.0.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (5.3.2)
Requirement already satisfied: rsa<5,>=3.1.4 in /isaac-sim/kit/python/lib/python3.10/site-packages (from google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (4.9)
Requirement already satisfied: requests-oauthlib>=0.7.0 in /isaac-sim/exts/omni.pip.cloud/pip_prebundle (from google-auth-oauthlib<2,>=0.5->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.3.1)
Requirement already satisfied: pyasn1<0.6.0,>=0.4.6 in /isaac-sim/kit/python/lib/python3.10/site-packages (from pyasn1-modules>=0.2.1->google-auth<3,>=1.6.3->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.5.1)
Requirement already satisfied: charset-normalizer<4,>=2 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from requests<3,>=2.21.0->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.1.1)
Requirement already satisfied: idna<4,>=2.5 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from requests<3,>=2.21.0->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.4)
Requirement already satisfied: certifi>=2017.4.17 in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from requests<3,>=2.21.0->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2023.5.7)
Requirement already satisfied: oauthlib>=3.0.0 in /isaac-sim/exts/omni.pip.cloud/pip_prebundle (from requests-oauthlib>=0.7.0->google-auth-oauthlib<2,>=0.5->tensorboard<3.0.0,>=2.8.0->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.2.2)
Requirement already satisfied: promise<3,>=2.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from wandb<0.13.0,>=0.12.11->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (2.3)
Requirement already satisfied: GitPython>=1.0.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from wandb<0.13.0,>=0.12.11->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (3.1.40)
Requirement already satisfied: docker-pycreds>=0.4.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from wandb<0.13.0,>=0.12.11->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.4.0)
Requirement already satisfied: pathtools in /isaac-sim/kit/exts/omni.kit.pip_archive/pip_prebundle (from wandb<0.13.0,>=0.12.11->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (0.1.2)
Requirement already satisfied: shortuuid>=0.5.0 in /isaac-sim/kit/python/lib/python3.10/site-packages (from wandb<0.13.0,>=0.12.11->rl-games==1.6.1->omniisaacgymenvs==2023.1.0b0) (1.0.11)
[... cut ...] tegers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, google.protobuf.pyext._message, psutil._psutil_linux, psutil._psutil_posix (total: 24)
/isaac-sim/python.sh: line 41:   198 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

Any help is appreciated

Toni-SM, 2023-12-15:
Hi @Cdfghglz 

Since you are working in a docker container, try passing the argument `headless=True` to the training script

Cdfghglz, 2023-12-19:
Thank you for the reply, I got further: 

```bash
root@xps15:/workspace/omniisaacgymenvs/omniisaacgymenvs# /isaac-sim/python.sh scripts/rlgames_train.py task=Cartpole headless=True
/isaac-sim/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.gym.headless.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window', '--allow-root']
Passing the following args to the base kit application:  ['task=Cartpole', 'headless=True']
Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/2023.1/user.config.json'
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2023.1/kit_20231219_075628.log
2023-12-19 07:56:28 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2023-12-19 07:56:28 [17ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad-200.0.0-rc.4+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [17ms] [Warning] [omni.ext.plugin] [ext: omni.kit.converter.cad_core-200.0.0-rc.3+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [19ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.core-103.4.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [20ms] [Warning] [omni.ext.plugin] [ext: omni.kit.sequencer.usd-103.4.2+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [23ms] [Warning] [omni.ext.plugin] [ext: omni.kit.widget.timeline-105.0.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [23ms] [Warning] [omni.ext.plugin] [ext: omni.kit.window.sequencer-103.4.2-dev.3+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [26ms] [Warning] [omni.ext.plugin] [ext: omni.paint.brush.attributes-1.3.1+105.0] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
2023-12-19 07:56:28 [28ms] [Warning] [omni.ext.plugin] [ext: omni.usd.schema.sequence-2.3.0+105.0.lx64.r.cp310] Built using kit version: 105.0. Current version: 105.1. It is considered compatible, but building with a newer version is recommended.
[0.061s] [ext: omni.kit.async_engine-0.0.0] startup
[0.463s] [ext: omni.assets.plugins-0.0.0] startup
[0.466s] [ext: omni.stats-0.0.0] startup
[0.467s] [ext: omni.client-1.0.1] startup
[0.478s] [ext: omni.gpu_foundation-0.0.0] startup
[0.496s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.498s] [ext: carb.windowing.plugins-1.0.0] startup
2023-12-19 07:56:28 [481ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2023-12-19 07:56:28 [481ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[0.499s] [ext: omni.kit.renderer.init-0.0.0] startup
2023-12-19 07:56:28 [506ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.129.03    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Laptop.. | Yes: 0 |     | 8188    MB | 10de      | 0          |
|     |                                  |        |     |            | 28a0      | 09391f11.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 6.0.12-76060012-generic
[... cut ...]
2023-12-19 07:58:47 [139,029ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2023-12-19 07:58:47 [139,029ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-12-19 07:58:47 [139,039ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-12-19 07:58:47 [139,049ms] [Warning] [omni.core.ITypeFactory] Module /isaac-sim/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
```

Toni-SM, 2023-12-19:
Now the problem is the RL training configuration.
Check the first point in [troubleshoot](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/blob/8cf773ab6cac0c8e0d55f46d6d7d258e781c6458/docs/troubleshoot.md#rl-training)

Note: your `num_envs` is 8 
  
```
2023-12-19 07:58:46 [138,814ms] [Error] [omni.kit.app._impl] [py stderr]: assert(self.batch_size % self.minibatch_size == 0)
```

Cdfghglz, 2023-12-19:
Thanks a lot @Toni-SM , changed the num_envs to 512 the example works nicely.

Would it be a worthy improvement to change the defaults to something that works out of the box or am I missing something about the config?

closed 2023-12-19 by Cdfghglz